TR4W/TR4W #168: ADD Indiana QP to contest selections of TR4W

n4af, 2016-05-18:
After using TR4W in general QSO mode in non-supported QSO parties  for years, I finally tried setting it up to work the four QP weekend on May 7...
 
I created an instate dom file that had all the 7QP, INQP, NEQP and the 3 DEQP counties, in numerous possible spellings... including the old MARAC abbreviations.... and I ran it under 7QP, as the exchange is the same for all the QPs.
 
 I also created an out of state county list with the 4 parties in it. That way, anybody could operate any or all of the QSO parties. Scoring was not possible as I am not that smart and not a programmer, but logging works. The bigger QPs have log checking that eliminates QSOs that don't count, and they score the log for you.
 
Not perfect, but fun to use. ...being an Indiana mobile I had to go back with a text editor and change my QTH from IN to my county, but just for the INQP entry. Fixed stations can enter their state (and county if in one of the 4 QP areas)
 
So my question is.....  Now that I have created the dom files for working the four QP weekend, how do we get it to be a supported QSO party (perhaps without scoring) so we can give it away on the INQP website?
 
Mel KJ9C

n4af, 2017-07-11:
last time i checked there was not a consensus on what the std exch would be in the 4 events.

ny4i, 2021-04-29:
I added this for this weekend. It is in the build _New_contest_IN7QPNE. The exchange has been standardized. The points for 7QP are different but the sponsors will calculate that. The one Cab file can be sent to all four sponsors.

n4af, 2021-04-29:
I see the new dom but where is contest array for IN7QPNE ?
Only file changed is addn of the dom ??

ny4i, 2021-04-29:
Haven’t checkin yet. Had to get back to work. Tonight. 

Principal Solutions Architect
Better Software Solutions, Inc. 
727-437-2771

n4af, 2021-04-29:
rr

n4af, 2021-04-29:
i can probably code the rest up...

ny4i, 2021-04-29:
I have it all done. I just need to reset so I can check it in again. I will get it by 6 or so tonight. But if you want to do it, here are the two files that changed besides the new DOM.

Tom
Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

n4af, 2021-04-29:
i put up 4.99.7
gonna see if kj9c can test it.

ny4i, 2021-04-29:
I’ll test it too. If it has the same files, it should work. Mine did here in testing.

Thanks,

Tom
Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

n4af, 2021-04-29:
i already had something coded. hopefully a close match ;-)

73, Howie

ny4i, 2021-04-29:
As long as you made the contest IN7QPE. That is the common name for the combined effort.

Tom
Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

n4af, 2021-04-29:
yup. hopefully ok in-state and outa state.

73, h

ny4i, 2021-04-29:
NDE, SDE nor KDE work though.

ny4i, 2021-04-29:
It works if I make the DOM file as follows:

```
NDE = NDE
KDE = KDE
SDE = SDE
INCLUDE FILE NEQSO.DOM
INCLUDE FILE IN_CTY.DOM
INCLUDE FILE SEVEN_CTY.DOM
```

I left out the states and provinces as out of state stations do not work them.

n4af, 2021-04-29:
which state are nde,kde,sde ?
Why are they not added to the appropriate cty file ?

ny4i, 2021-04-29:
Delaware. Delaware does not have a separate one so it is in the IN7QPNE.DOM

Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

n4af, 2021-04-29:
Hi Tom -

Would you download <https://tr4w.net/4.99/tr4w_setup_4_99.7.exe>/test again
?

Tks, Howie

ny4i, 2021-04-29:
Looks good. I tested several counties.

```
QSO:  1800 PH 2021-04-29 2110 NY4I          59  FL     W1AW       59  HARCT        
QSO:  1800 PH 2021-04-29 2110 NY4I          59  FL     W7SP       59  UTSAL        
QSO:  1800 PH 2021-04-29 2121 NY4I          59  FL     W3DE       59  KDE          
QSO:  1800 CW 2021-04-29 2350 NY4I          599 FL     W7REF      599 AZMHV        
QSO:  1800 CW 2021-04-29 2351 NY4I          599 FL     W1AE       599 BRIRI        
QSO:  1800 CW 2021-04-29 2351 NY4I          599 FL     W7ID       599 IDONE        
QSO:  1800 CW 2021-04-29 2351 NY4I          599 FL     W3WS       599 NDE          
QSO:  1800 CW 2021-04-29 2351 NY4I          599 FL     W1ME       599 YORME        
QSO:  1800 CW 2021-04-29 2352 NY4I          599 FL     W7UYU      599 UTUTA        
QSO:  1800 CW 2021-04-29 2352 NY4I          599 FL     W3CC       599 SDE          
QSO:  1800 CW 2021-04-29 2352 NY4I          599 FL     W3IU       599 KDE          
QSO:  1800 CW 2021-04-29 2352 NY4I          599 FL     W9IU       599 INVER   
```

n4af, 2021-04-30:
Ok. Tks Tom...

closed 2022-02-07 by ny4i