TR4W/TR4W #181: EUHFC Truncates Leading Zero

n4af, 2016-08-12:
TR4W support EUHFC. EUHFC rules saz that
echange is 2 digit number of the license. We are getting Cabillo logs with
just one number from TR4W Cabrillo, ledaing yeros are not there but they
should be. A soon I type licence year 09 in TR4W it is logged as 9 only, it
is not 2 digit what we would like to have.

Cabrillo looks like that:

QSO:  1800 CW 2016-08-12 0959 S57AW         599 87     S59AA         599 50

QSO:   1800 CW 2016-08-12 0959 S57AW         599 87     S52RE         599 9
 -  must be  599 09

QSO:  1800 CW 2016-08-12 0959 S57AW         599 87     UW2GG         599 0
-  must be  599 00

QSO:  1800 CW 2016-08-12 1000 S57AW         599 87     EA3BO         599 89

QSO:  1800 CW 2016-08-12 1000 S57AW         599 87     HB9AAS        599 15

QSO:  1800 CW 2016-08-12 1000 S57AW         599 87     LA0GB         599 2
-  must be  599 02

Is there need to update TR4W or kind of settings. We woul like to have
Cabrillo with 00, 02, 09 etc... exchange

73 Robert, S57AW

n4af, 2016-08-12:
I think EUHFC handles 'year' by treating it as 'zone' and, hence, dropping the zero.

s57aw, 2016-08-12:
Tnx Howie and Val... Yes, I assumed the same thing - years treated as zones and without leading zeros. Contest was last weekend and see from uploaded logs that years (00 - 09) are without leading zeros and this is unique for TR4W. I am sure you can change this for 2017 contest... Plenty of time:-)

73 & GL de Robert, S57AW

n4af, 2016-08-12:
Fixed 4.51.1
Print all zones as two digits in post processing module PostUnit

n4af, 2016-08-12:
closing on postive feedback from UR7QM

closed 2016-08-12 by n4af