TR4W/TR4W #211: ADIF Import into contest shifts freqency

ny4i, 2016-11-28:
I imported  ADIF data into a new contest, but the frequency is shifted.

Here is a sample of what I imported:
```
<Band:3>40M <Call:7>TK/S57C <APP_DXKEEPER_DXCCPREFIX:2>TK <DXCC:3>214 <Freq:5>7.011 <Mode:2>CW <Name:6>Branko <Operator:4>NY4I <QSL_Via:13>BUREAU, e-QSL <QSO_DATE:8>20161128 <TIME_OFF:6>021837 <TIME_ON:6>021837 <QTH:5>Lumio <RST_Rcvd:3>599 <RST_Sent:3>599 <TX_PWR:5>500.0 <CONT:2>EU <ITUZ:2>28 <CQZ:2>15 <GRIDSQUARE:6>JN42jo <ADDRESS:32>Branko Zemljak

Lumio
CORSICA <PFX:3>TK0 <STATION_CALLSIGN:4>NY4I <OWNER_CALLSIGN:4>NY4I <LAT:11>N042 36.250 <LON:11>E008 49.283 <EMAIL:16>s57c@hamradio.si <EQSL_QSL_SENT:1>R <LOTW_QSL_SENT:1>Y <LOTW_QSL_RCVD:1>N <LOTW_QSLSDATE:8>20161128 <APP_DXKeeper_ClubLog:1>Y <APP_DXKeeper_ClubLogDate:19>2016-11-28 16:37:13 <APP_DXKEEPER_MY_QTHID:15>W1/EastportMult <EOR>
<Band:4>160M <Call:8>TK/S53CC <APP_DXKEEPER_DXCCPREFIX:2>TK <DXCC:3>214 <Freq:5>1.832 <Mode:2>CW <Name:5>Boris <Operator:4>NY4I <QSO_DATE:8>20161128 <TIME_OFF:6>021750 <TIME_ON:6>021750 <RST_Rcvd:3>599 <RST_Sent:3>599 <TX_PWR:6>1500.0 <CONT:2>EU <ITUZ:2>28 <CQZ:2>15 <ADDRESS:50>Boris Hrovat
Podrecje 53A
1230 Domzale
SLOVENIA <PFX:3>TK0 <STATION_CALLSIGN:4>NY4I <OWNER_CALLSIGN:4>NY4I <EQSL_QSL_SENT:1>R <LOTW_QSL_SENT:1>Y <LOTW_QSL_RCVD:1>N <LOTW_QSLSDATE:8>20161128 <APP_DXKeeper_ClubLog:1>Y <APP_DXKeeper_ClubLogDate:19>2016-11-28 16:37:13 <APP_DXKEEPER_MY_QTHID:12>W1/Calais160 <EOR>
<Band:3>80M <Call:4>TK0C <APP_DXKEEPER_DXCCPREFIX:2>TK <DXCC:3>214 <Freq:6>3.5307 <Mode:2>CW <Name:7>Corsica <Operator:4>NY4I <QSL_Via:35>LOTW, EQSL buro and direct via S50C <QSO_DATE:8>20161127 <TIME_OFF:6>234609 <TIME_ON:6>234609 <QTH:14>Sant Ambroggio <RST_Rcvd:3>599 <RST_Sent:3>599 <TX_PWR:5>500.0 <CONT:2>EU <ITUZ:2>28 <CQZ:2>15 <CONTEST_ID:8>CQ-WW-CW <ADDRESS:66>Corsica Dxpedition - Qth Sant Ambroggio

Sant Ambroggio
CORSICA <PFX:3>TK0 <STATION_CALLSIGN:4>NY4I <OWNER_CALLSIGN:4>NY4I <APP_DXKeeper_LotW_MEMBER:1>Y <EQSL_QSL_SENT:1>R <LOTW_QSL_SENT:1>Y <LOTW_QSL_RCVD:1>N <LOTW_QSLSDATE:8>20161128 <APP_DXKeeper_ClubLog:1>Y <APP_DXKeeper_ClubLogDate:19>2016-11-28 00:10:34 <APP_DXKEEPER_MY_QTHID:9>W1/Calais <EOR>
<Band:3>80M <Call:6>SV2DSJ <APP_DXKEEPER_DXCCPREFIX:2>SV <DXCC:3>236 <Freq:5>3.584 <Mode:2>CW <Name:10>Panagiotis <Operator:4>NY4I <QSL_Via:18>Only DIRECT - LoTW <QSO_DATE:8>20161127 <TIME_OFF:6>234504 <TIME_ON:6>234504 <QTH:9>Kalamaria <RST_Rcvd:3>599 <RST_Sent:3>599 <TX_PWR:5>500.0 <CONT:2>EU <ITUZ:2>28 <CQZ:2>20 <CONTEST_ID:8>CQ-WW-CW <GRIDSQUARE:6>KN10ln <ADDRESS:80>Panagiotis Violistis
18 Stroggili Str.
Kalamaria, THESSALONIKI, 551 32
GREECE <PFX:3>SV2 <STATION_CALLSIGN:4>NY4I <OWNER_CALLSIGN:4>NY4I <LAT:11>N040 34.467 <LON:11>E022 57.950 <EMAIL:16>sv2dsj@gmail.com <APP_DXKeeper_LotW_MEMBER:1>Y <EQSL_QSL_SENT:1>R <LOTW_QSL_SENT:1>Y <LOTW_QSL_RCVD:1>Y <LOTW_QSLSDATE:8>20161128 <LOTW_QSLRDATE:8>20161128 <APP_DXKeeper_ClubLog:1>M <APP_DXKeeper_ClubLogDate:19>2016-11-28 00:10:34 <APP_DXKeeper_LotWConfirmation:2>GM <APP_DXKEEPER_MY_QTHID:9>W1/Calais <EOR>
<Band:4>160M <Call:4>V26K <APP_DXKEEPER_DXCCPREFIX:2>V2 <DXCC:2>94 <Freq:6>1.8451 <Mode:2>CW <Name:6>Joseph <Operator:4>NY4I <QSL_Via:60>AA3B (Direct or Bureau) + LOTW, eQSL, QRZ.com, Club Log OQRS <QSO_DATE:8>20161127 <TIME_OFF:6>234230 <TIME_ON:6>234230 <QTH:9>Boyertown <RST_Rcvd:3>599 <RST_Sent:3>599 <TX_PWR:6>1500.0 <CONT:2>NA <ITUZ:2>11 <CQZ:1>8 <IOTA:6>NA-100 <CONTEST_ID:8>CQ-WW-CW <ADDRESS:47>Joseph Trench
10 Senia Lane
Boyertown,  
USA <PFX:3>V26 <CNTY:5>Berks <STATION_CALLSIGN:4>NY4I <OWNER_CALLSIGN:4>NY4I <EMAIL:18>aa3b.bud@gmail.com <APP_DXKeeper_LotW_MEMBER:1>Y <EQSL_QSL_SENT:1>R <LOTW_QSL_SENT:1>Y <LOTW_QSL_RCVD:1>N <LOTW_QSLSDATE:8>20161128 <APP_DXKeeper_ClubLog:1>Y <APP_DXKeeper_ClubLogDate:19>2016-11-28 00:10:34 <APP_DXKEEPER_MY_QTHID:12>W2/Summit160 <EOR>
<Band:4>160M <Call:4>CN2R <APP_DXKEEPER_DXCCPREFIX:2>CN <DXCC:3>446 <Freq:6>1.8271 <Mode:2>CW <Name:3>Jim <Operator:4>NY4I <QSL_Via:4>W7EJ <QSO_DATE:8>20161127 <TIME_OFF:6>233926 <TIME_ON:6>233926 <QTH:1>. <RST_Rcvd:3>599 <RST_Sent:3>599 <TX_PWR:6>1500.0 <CONT:2>AF <ITUZ:2>37 <CQZ:2>33 <CONTEST_ID:8>CQ-WW-CW <ADDRESS:34>Jim Sullivan
Via W7ej
.
MOROCCO <PFX:3>CN2 <STATION_CALLSIGN:4>NY4I <OWNER_CALLSIGN:4>NY4I <EMAIL:13>cn2r@cn2r.net <EQSL_QSL_SENT:1>R <LOTW_QSL_SENT:1>Y <LOTW_QSL_RCVD:1>N <LOTW_QSLSDATE:8>20161128 <APP_DXKeeper_ClubLog:1>Y <APP_DXKeeper_ClubLogDate:19>2016-11-28 00:10:34 <APP_DXKEEPER_MY_QTHID:12>W2/Summit160 <EOR>
<Band:3>20M <Call:6>XX9TXN <APP_DXKEEPER_DXCCPREFIX:3>XX9 <DXCC:3>152 <Freq:7>14.0448 <Mode:2>CW <Name:7>Alberto <Operator:4>NY4I <QSL_Via:6>VR2XAN <QSO_DATE:8>20161127 <TIME_OFF:6>233124 <TIME_ON:6>233124 <QTH:9>Peng Chau <RST_Rcvd:3>599 <RST_Sent:3>599 <TX_PWR:5>500.0 <CONT:2>AS <ITUZ:2>44 <CQZ:2>24 <IOTA:6>AS-075 <CONTEST_ID:8>CQ-WW-CW <ADDRESS:58>Alberto Annesi
1/F 7A Nam Shan Road
Peng Chau
HONG KONG <PFX:3>XX9 <STATION_CALLSIGN:4>NY4I <OWNER_CALLSIGN:4>NY4I <EQSL_QSL_SENT:1>R <LOTW_QSL_SENT:1>Y <LOTW_QSL_RCVD:1>N <LOTW_QSLSDATE:8>20161128 <APP_DXKeeper_ClubLog:1>Y <APP_DXKeeper_ClubLogDate:19>2016-11-28 00:10:34 <APP_DXKEEPER_MY_QTHID:9>W7/Tacoma <EOR>
<Band:4>160M <Call:5>CN2AA <APP_DXKEEPER_DXCCPREFIX:2>CN <DXCC:3>446 <Freq:6>1.8339 <Mode:2>CW <Name:4>Rw2f <Operator:4>NY4I <QSL_Via:5>UA2FM <QSO_DATE:8>20161127 <TIME_OFF:6>231745 <TIME_ON:6>231745 <QTH:8>nr Rabat <RST_Rcvd:3>599 <RST_Sent:3>599 <TX_PWR:6>1500.0 <CONT:2>AF <ITUZ:2>37 <CQZ:2>33 <CONTEST_ID:8>CQ-WW-CW <GRIDSQUARE:6>IM64pd <ADDRESS:49>Rw2f &Amp; Rl3a Contest Team
```

This is what the Cabrillo file I then exported has:
...
QSO:  7011 CW 2016-11-28 0218 NY4I          599 5      TK/S57C       599 15           
QSO:  1832 CW 2016-11-28 0217 NY4I          599 5      TK/S53CC      599 15           
QSO:    35 CW 2016-11-27 2346 NY4I          599 5      TK0C          599 15           
QSO:  3584 CW 2016-11-27 2345 NY4I          599 5      SV2DSJ        599 20           
QSO: 18451 CW 2016-11-27 2342 NY4I          599 5      V26K          599 8            
QSO: 18271 CW 2016-11-27 2339 NY4I          599 5      CN2R          599 33           
QSO:   140 CW 2016-11-27 2331 NY4I          599 5      XX9TXN        599 24           
QSO: 18339 CW 2016-11-27 2317 NY4I          599 5      CN2AA         599 33    
...

According to the ADIF spec, the frequency is in Mhz. So, the embedded . is correct.

n4af, 2016-11-29:
Just to document how TR4W is designed, it expects <FREQ> to be in hertz on IMPORT.
It should then EXPORT in Khz.
The decimal point on input is thrown away.
So ...

ADIF <FREQ:7>3535450
 3535450 input - or 3.535450- will produce 3535.45

<FREQ:8>7.038330  should import as 7038.33
note the above would be the same for
<FREQ:7>7038330

Works internally, but it would be nice to enhance TR4W to import from other products using the decimal point as a delimiter.

ny4i, 2016-11-29:
> Just to document how TR4W is designed, it expects to be in hertz on IMPORT.

Howie,

I recalled some of that from when I looked at this before. I will take this one. It will be a nice way to get back into things.

Tom
Principal Solutions Architect, ITIL®v3 
Better Software Solutions, Inc.
727-437-BSS1 (727-437-2771)
@BSSI_Consulting

n4af, 2016-11-29:
Thanks Tom -
MainUnit procedure ImportFromADIF needs to parse the decimal point he throws away ;-)

73, Howie

n4af, 2016-12-19:
DXKeeper needs to terminate each record with <EOR> else it will not write out to the log.
http://www.adif.org/adif.html
73, Howie

ny4i, 2016-12-19:
> DXKeeper needs to terminate each record with else it will not write out to the log.
> http://www.adif.org/adif.html <http://www.adif.org/adif.html>
> 73, Howie

Howie,

Did you leave out a word? :)

I ask as I just exported some records from DXKeeper and it terminates each record with an <EOR>

Tom
Principal Solutions Architect, ITIL®v3 
Better Software Solutions, Inc.
727-437-BSS1 (727-437-2771)
@BSSI_Consulting

n4af, 2016-12-19:
Ok. I must have missed seeing the EOR.
I am looking at N1MM, where there can be an import problem.

73, Howie

n4af, 2016-12-20:
Ok. I might have missed the <EOR>'s
Using the one line

n4af, 2016-12-20:
Hi Tom -

I may see a way to be more universal.
Set var PosDecimal to the integer value of the position the '.' was found in ADIF_FREQ

At the end of ADIF_FREQ processing, insert the following lines:

if PosDecimal >0 then
          begin
            while FieldLength - PosDecimal < 8 do          // 4.55.1
              begin
                TempRXData.Frequency := TempRXData.Frequency * 10;
                FieldLength := FieldLength + 1 ;
              end;
            end;

Could you run http://tr4w.net/beta/tr4w.exe
against some of your logkeeper ADIF and see if it works ?
If there are still some problems, maybe it will give you a starting off point ??

73, Howie

n4af, 2016-12-20:
if (PInteger(@TempBuffer)^ = ADIF_FREQ) then
    begin
      FieldLength := GetNumberFromCharBuffer(@CurrentPos[5]);
      again:
      for TempInteger := 0 to FieldLength-1  do
        begin
         if StartPos[TempInteger] = '.' then
          begin
          PosDecimal := TempInteger;
            if Switch = False  then
              begin
               FieldLength :=   FieldLength + 1;
               Switch := True;
                  //    goto again;
              end;
           end;
         if StartPos[TempInteger] in ['0'..'9'] then
           TempRXData.Frequency := Ord(StartPos[TempInteger]) - Ord('0') + TempRXData.Frequency * 10
           else
            if StartPos[TempInteger] = '' then
             TempRXData.Frequency := TempRXData.Frequency * 10;
         end;
         if PosDecimal >0 then
          begin
            while FieldLength - PosDecimal < 8 do          // 4.55.1
              begin
                TempRXData.Frequency := TempRXData.Frequency * 10;
                FieldLength := FieldLength + 1 ;
              end;
            end;

ny4i, 2016-12-20:
The frequency looks OK. Importing contests is odd as one has to address the context. For example, I exported some records from STEW-PERRY from DXKeeper. I exported them to an ADIF file. I imported the file in TR4W (your beta) and the freq was OK but the QTH was the state and County. The Grid square is in the ADIF file but TR does not apply the logic of the contest against what it imports (ie, realize that SP cares are grid and put the grid int he QTH).  This may be far too more than what we need to do though. Certainly exporting ADIF is the thing that happens more :)

I did a test with N1MM to see what it did when I imported the same ADIF. It does appear to search out the grid in the ADIF file. 

I’m happy with the frequency being right. 

Thanks,

Tom
Principal Solutions Architect, ITIL®v3 
Better Software Solutions, Inc.
727-437-BSS1 (727-437-2771)
@BSSI_Consulting

n4af, 2016-12-20:
FB Tom -

Yes, I think a lot of the fields only load to specifics of the contest.

I will see if any of the support group can beat on it.

4.55.1

73, Howie

n4af, 2017-10-17:
closing this as I think frequency is working as designed.

closed 2017-10-17 by n4af