TR4W/TR4W #218: Add support for FT991, 991A and 891

ny4i, 2016-12-28:
Some commands (FA for example) have changed. Identify the changes and code up a new type of Yaesu with the longer FA field and whatever other changes are found.

ny4i, 2016-12-28:
This was a good segue to get me back to coding. Plus I introduced a bug when I put 991 in the list but it would not work. This is actually easier to do that I thought as I'm almost done. We will of course need to have Yaesu rig owners test it. Preferably someone with an 891 and an older Yaesu to make sure I did not break anything.

n4af, 2016-12-28:
Thanks Tom !!!!

ny4i, 2016-12-29:
User reports that this now works on the FT891

ny4i, 2016-12-30:
I forgot to do a Global search for a radio to make sure I added FT891 and FT991 to every place where a Yaesu rig was specified. I found two more commands that had something specific for the radio (UP and DN command). This fixes that.

commit b4ba8b7c (2020-07-31): Update Split and TXOn code plus Debug

Also added a debug file facility. This closes #218, #138 and #147.
closed 2020-07-31 by n4af