TR4W/TR4W #244: New propositions for YU-DX CONTEST

yu1lg, 2017-03-09:
Dear all,
 
Please find attached to this letter the new propositions for YU-DX CONTEST.
 
The term to hold the contest is still the third full weekend of April, i.e. Saturday at 12:00 UT up to Sunday at 11:59 UT (15/16th of April 2017).
 
[INTERNATIONAL_YUDX_HF_CONTEST.pdf (English)](https://github.com/n4af/TR4W/files/830465/INTERNATIONAL_YUDX_HF_CONTEST.pdf)

[YUDXC 2017.pdf (Serbian)](https://github.com/n4af/TR4W/files/830466/YUDXC.2017.pdf)

n4af, 2017-03-10:
ADDED 4.57.5

yu1lg, 2017-03-11:
Thank you very much

n4af, 2017-03-12:
change mults so YU not awarded domestic mults:
"
Multiplier for non-YU stations:

The sum of worked DXCC countries + worked YU/YT COUNTIES on each band, regardless of the mode

Multiplier for YU stations:
The sum of worked DXCC countries on each band, regardless of the mode
"
added 4.57.7

n4af, 2017-03-12:
changed code to allow YU to count as a Country.

n4af, 2017-03-12:
revise YU to YU QsoPoints to 1

n4af, 2017-03-12:
add yugoslav county full names to abbreviations yu.dom

n4af, 2017-03-12:
note: -for YU stations- county entered is WYSIWYG

n4af, 2017-04-19:
oops, this is already in YUDX.

closed 2017-04-19 by n4af