TR4W/TR4W #259: CTRL-ALT-R accelerator not working to turn rotator

ny4i, 2017-05-28:
The menu option works but nothing happens wen pressing the CTRL-ALT-R combo.

n4af, 2023-08-26:
ctrl-alt-r is a windows shortcut and not available to TR4W.

4.125.4 assigns ctrl-p to Redo possible calls- if no call is entered. then rotor control is invoked.

n4af, 2023-08-26:
This removes ctrl-P PTT activation and assigns to dual function redo possible calls/rotor control.

closed 2023-08-26 by n4af

ny4i, 2023-08-26:
If no call is entered, where does ALT-P turn the rotor to? I thought the idea was to turn it to the estimated location of the call in the call window? Principal Solutions ArchitectBetter Software Solutions, Inc. 727-437-2771

n4af, 2023-08-26:
Hi Tom -

Well, if no call is entered, trace shows him dropping thru
VisibleLog.DoPossibleCalls(CallWindowString);
on the way to
 RotorControl(LastHeadingShown);
which was previously set by
***@***.***);
resulting in no motion as rotor control is already set to the prev beam
heading ??

Are you thinking this should be titled something like 'beam heading'
instead of 'redo possible calls' ?

Thanks, Howie

n4af, 2023-08-26:
of course, if ctl-j rotor parms are not set, we do not get past
***@***.***);
        DisplayGridSquareStatus(CallWindowString);
        VisibleLog.DoPossibleCalls(CallWindowString);

ny4i, 2023-08-26:
No. Just asking. I admit I don’t know what REDO POSSIBLE CALLS does. I’ve never looked. Principal Solutions ArchitectBetter Software Solutions, Inc. 727-437-2771

n4af, 2023-08-27:
i think i will rename it, redo seems to just reload possible calls.

Tks, Howie