TR4W/TR4W #260: Strange VirtualStorage issue

ny4i, 2017-05-28:
If the Windows system is using VirtualStorage, when doing an export, the file is displayed but if one selects Open With Notepad, the file is not found.

This requires more research but TR4W can detect if the VirtualStore is on use (set by LocalPolcy) and look in the right place.

ny4i, 2017-11-24:
Some more information...In ARRL SS SSB, I selected export to Cabrillo, and I see the window with c:\Program Files (x86)\TR4W\2017 ARRL-SS-SSB NY4I\NY4I.LOG int he title bar. However, if I select Open with Notepad, I receive a message that the system could not find the path specified. 

Also, the directory 2017 ARRL-SS-SSB NY4I\ does not exist in c:\Program Files (x86)\TR4W\. 

If I do a search for NY4I.LOG on the entire C drive, I find it in c:\Users\toms\AppData\Local\VirtualStore\Program Files (x86)\TR4W\2017 ARRL-SS-SSB NY4I.

It seems that in some cases, Windows 7 (in this case) is using a Virtual Store. That may not be an issue but the name of the file is wrong int he heading and however the program is building the file name to pass to Notepad is wrong.
[image]
[image]
[image]

n4af, 2017-11-24:
Virtual storage seems a common complaint w/ windows 7.
You may be able to disable it via this posting:

https://answers.microsoft.com/en-us/windows/forum/windows_7-security/how-to-disable-virtualstore-in-windows-7/55dce284-0dcd-46af-892e-d2b9cf5bcff6?auth=1

ny4i, 2017-11-24:
I went through that procedure but the take Ownership trick did not work.

I believe we either need to handle the Virtual Store properly when we export the files (and offer up Notepad) or we need to start recommending people install the software in c:\TR4W.

There may be another way to tell Windows to not use the virtual store when we create the files. More research.

It is a bit odd to see a display that indicates the directory is in c:\Program Files (x86)\TR4W but when you look there with explorer, it is not there.

closed 2026-11-04 by ny4i