TR4W/TR4W #271: PLS ADD RSGB DX Contest

n4af, 2017-07-26:
Please add the new RSGB DX contest:
http://www.rsgbcc.org/hf/rules/2017/rOctoberDX.shtml

n4af, 2023-06-08:
Ok. So RSGBDX will be removed.

Did you have a problem with  Callsign Update Enable ?
I would have thought that whitespace between serial # and district code would not cause TR4W to interpret that as a callsign ?

n4af, 2023-06-08:
I have removed RSGBDX. 
The exchange is automatically set for 
 case ActiveExchange of
    RSTQSONumberExchange: SetUpRSTQSONumberExchange;
This is not the case for UKEI as stations residing within UK/EI will add their two character district code.
However, I will add code to check if UKEI and, if not, add the RST QSO NUMBER exchanges.

df5en, 2023-06-08:
I tested this:

1.
CQ CW EXCHANGE=5NN # (no space between "=" and "5NN")
CQ CW EXCHANGE= 5NN # (one space between "=" and "5NN")
CQ CW EXCHANGE=  5NN # (two spaces between "=" and "5NN")

In all cases the following is sent in CQ mode with the call G1ABC: "G1ABC5NN 001". This would irritate G1ABC as he might not understand his rapport.(no space between "=" and "5NN")

2.
With CQ CW EXCHANGE=_5NN # the string "G1ABC 5NN 001" (1 space between G1ABC and 5NN) is sent correctly. A single underscore "_" works.

3.(no space between "=" and "5NN")
With CQ CW EXCHANGE=__5NN # the string "G1ABC  5NN 001" (2 spaces between G1ABC and 5NN) is sent. Actually, the break between G1ABC and 5NN is a bit too long here. Well, I felt it was just right for the contest in April, today I think it's too long.

n4af, 2023-06-08:
I am not sure why you are doing this ?
Normally, ALT-P then 'O' then fill in the prompts.
The result is written to the config file.
I have done hundreds of contests over the years with TR4W w/o trouble.

Typically, ALT-P is the user interface for customizing the exchanges.

closed 2026-03-14 by ny4i