TR4W/TR4W #282: Add Icom 7610 as a radio (CI-V Address 98h)

ny4i, 2017-10-25:
The Japanese manual stated the address to be 98h. Verify the basic operation to be just as the 7610 and then add the new radio type with address 98h.

ny4i, 2017-11-24:
I compared the IC7610 Japanese Supplementary manual to the English 7300 manual. The functions that TR4W uses are identical so rather than make the 7610 just like the 7600, change it to be in the same category as the 7300 code in the radio definition arrays and all should be ok. I will dig out the dev laptop (have to do that for a UDP fix for Marv) and get that coded.

n4af, 2017-11-27:
RR. Will do. Tks, Howie

n4af, 2017-11-27:
closed - build 4.65.7 adds type ic7610

ny4i, 2017-11-27:
Howie, did you add it to the arrays of radio’s that support CWByCAT and
CWSpeed?

Thanks

Tom

Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

ny4i, 2017-11-27:
I guess I should change that to add a CWByCAT option to the radio
properties array. I’m not sure why I didn’t do it that way.

Tom

Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

n4af, 2017-11-27:
Ok. The changes for CWBC and speed are in my master and at
https://github.com/n4af/TR4W/commit/0bd44e635a0abb220e158130758b2e10c60fddca

They will not actually make code until the next mod 4.65.8 or 4.66

73, Howie

ny4i, 2018-02-16:
Howie, It looks like the IC7610 entry was simply a clone of the line for the IC7600. The address looks off as it is the same address as the IC7600. The $7A below should be $98 for the 7610.

 ({Name: 'IC7610';     } BR:BR9600;  P: 1; C: 0; S: 0; T:0; RA: **$7A;** SW: 1; SFOC: $00; SMOC: $00; SPOC: $00; RX: $00; TX: $00; MB: 0; CW: $00; LSB: $00; USB: $00; FM: $00; AM: $00; DIGL: $00; DIGU: $00; rt: rtIcom),    // 4.65.7  issue #282

I will check the code to see what is happening elsewhere.

n4af, 2018-02-16:
Hi Tom

I think this <http://tr4w.net/4.68/tr4w_setup_4_68.12.exe>reflects the
change u suggested.

to:
({Name: 'IC7610';     } BR:BR9600;  P: 1; C: 0; S: 0; T:0; RA: *$98*; SW:
1; SFOC: $00; SMOC: $00; SPOC: $00; RX: $00; TX: $00; MB: 0; CW: $00; LSB:
$00; USB: $00; FM: $00; AM: $00; DIGL: $00; DIGU: $00; rt: rtIcom), //
4.65.7 issue #282 #282   // 4.68.12

I will not be available until Monday, as in ARRLDX-CW prep mode.

73, Howie

ny4i, 2018-02-16:
Thanks. I will test that against my 7300 as it should be the same. I think Terry might have some other issue.

GL in the contest. I have no doubt some will work me and think they worked you and vice versa (If you are using NY4A) :)

Tom
Principal Solutions Architect, ITIL®v3 
Better Software Solutions, Inc.
727-437-BSS1 (727-437-2771)
@BSSI_Consulting

ny4i, 2018-03-07:
I verified the default address works. I set my 7300 to use address 98h and I selected IC7610 in TR4W and it started communicating with the radio.

I will verify the commands sets again are identical for the functions we use.

n4af, 2018-03-07:
Outstanding. !
Thanks Tom

ny4i, 2018-03-07:
I did find a change in the 7610 from the 7300 so I am checking that out now. There will be a commit for this.

Tom
Principal Solutions Architect, ITIL®v3 
Better Software Solutions, Inc.
727-437-BSS1 (727-437-2771)
@BSSI_Consulting

ny4i, 2020-07-22:
This looks to be just like the 7300 except the extended mode command ($26) has a PKT mode and Multiple data modes. I wish I documented the difference I found so I will have to look closer.
Extended Mode Commands
00: LSB
01: USB
02: AM
03: CW
04: RTTY
05: FM 
07: CW-R 
08: RTTY-R 
12: PSK 
08: PSK-R

00: Data mode OFF
01: Data mode 1 (D1)
02: Data mode 2 (D2)
03: Data mode 3 (D3)

What are the three Data modes?

01: FIL1
02: FIL2
03: FIL3

ny4i, 2022-02-05:
This appears fixed. I asked a local friend with an IC-7610 to check the PSK mode reflects as DATA in the program.

closed 2022-02-06 by ny4i