TR4W/TR4W #289: Cannot find default log backup file

ny4i, 2017-11-24:
Upon first installing the program, the BACKUP LOG FILE NAME is set to logback.tr4w. If I want to change the drive that file is on, I double-click the value and am presented with an Open file dialog. I navigate to the directory I want to save the file (in my case DropBox). The issue is that upon first setting this up, the file logback.tr4w is not in the directory so I cannot open it.

This dialog needs to allow selecting a directory without a file and then the first backup would create the log backup.

I recommend testing this by selecting a directory without a logback.tr4w file to duplicate this behavior. I am curious if anyone else confirms this when the file is not there already.

n4af, 2017-11-27:
I am not sure what the win api call for opening a new file for write is.

not a perfect solution, but
what you can do is locate the line- or create it, in TR4W.INI in settings.
BACKUP LOG FILE NAME=E:\logbak.tr4w

This can be a new/non-existing file.

73, Howie

ny4i, 2017-11-27:
I think the issue is the flags that are set in the dialog.

At line 432 in JCTRL2.PAS, this seems to be the code to show the file dialog.

if OpenFileDlg(TC_SELECTNEWFLOPPYFILESAVENAME, settingswindowhandle, nil, TempString, OpenMMTTYFlags)
        then FloppyFileSaveName := TempString;

In uDialogs.pas, I see OpenMMTTYFlags defined to be the following:

OpenMMTTYFlags                        = OFN_ENABLESIZING or OFN_NOREADONLYRETURN or OFN_EXPLORER or OFN_FILEMUSTEXIST or OFN_PATHMUSTEXIS

Since the OFN_FILEMUSTEXIST flag is set, this does not allow the dialog to return a new filename.

I suspect if we set the dialog flags to allow a new file (my guess by using similar flags but without OFN_FILEMUSTEXIST), then you would get the name back as the user typed it.

Tom
Principal Solutions Architect, ITIL®v3 
Better Software Solutions, Inc.
727-437-BSS1 (727-437-2771)
@BSSI_Consulting

n4af, 2017-11-27:
> 727-437-BSS1 (727-437-2771 <(727)%20437-2771>)

Yes,
function OpenFileDlg in uDialogs is the culprit.

I can comment out line 892 to read
 ofn.Flags := Flags { or OFN_FILEMUSTEXIST} ;

but cannot be 100% sure it will not impact. ADIF import.

I think adif import is the only other client, so perhaps REMOVE
OFN_FILEMUSTEXIST from the dialog and adding to adif import:

In MainUnit.PAS routine  *menu_import_adif*:
if OpenFileDlg(nil, tr4whandle, 'ADIF (*.adi)'#0'*.adi',
TR4W_ADIF_FILENAME, OFN_HIDEREADONLY or OFN_ENABLESIZING)
to
if OpenFileDlg(nil, tr4whandle, 'ADIF (*.adi)'#0'*.adi',
TR4W_ADIF_FILENAME, OFN_HIDEREADONLY or OFN_ENABLESIZING *or
OFN_FILEMUSTEXIST*)

n4af, 2017-11-27:
in reflecting on this, i think my last stmt makes the most sense.
Namely let the callers set the flags. Why should the open dialog be setting a flag for
file must exist ?

1) remove file must exist from OpenFileDialog
2) use explicit flags in callers to #1

n4af, 2017-11-28:
This is going out for testing as 4.65.8

ny4i, 2017-11-28:
That makes sense.

Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

n4af, 2017-12-03:
closing- this is picked up on 4.66.0

closed 2017-12-03 by n4af

ny4i, 2017-12-03:
It works well. Thanks

Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771