TR4W/TR4W #290: TR4W CAT

S52CQ, 2017-11-27:
I am using open source mcHF radio which I am very pleased with.

It's software supports also CAT control, but authors of the SW have pointed out problem with TR4W. See the thread and description of the problem here:

https://github.com/df8oe/UHSDR/issues/1174

I am pleased with TR4W however I cannot use it if it does not fully function.

Can you please check the issue and see if TR4W can be upgraded?

73 de Jure, S52CQ

n4af, 2017-12-03:
TR4W does not support this.

S52CQ, 2018-01-24:
Hi Tony,

I've just tried it - it does work,  but only once. So at the power up it would not work, so I went to Setting/CAT & CW Keying>Radio 1. 
There I have settings:
Keyer Radio One Output Port: COM3
Radio One Keyer RTS: PTT
Radio One Keyer DTR: CW

I have "changed" Radio One Keyer RTS: from PTT to PTT, clicked Apply/OK and hit F9 (shortcut for ?)
Radio was keyed correctly only once, next time it would do nothing. If I change Radio One Keyer RTS: PTT again then F9 is keyed correctly, but only once.

Does that report help at locating the problem?

73s de S52CQ, Jure

ny4i, 2018-01-24:
Did you see the PTT VIA command?

Tom

Principal Solutions Architect, ITIL®v3 
Better Software Solutions, Inc.
727-437-BSS1 (727-437-2771)
@BSSI_Consulting

S52CQ, 2018-01-24:
Yes, I've followed your instruction.

73 de S52CQ, Jure

ny4i, 2018-01-24:
OK. I asked as you gave me a summary of what you did, but I did not read any mention of setting PTT VIA. One last item…

set Radio One Keyer RTS: PTT to None instead of PTT. You are using a CAT command so RTS is not involved in this…

I have an FT-817 but as I am moving next week, I will have to revisit this the second week of February.

Tom
Principal Solutions Architect, ITIL®v3 
Better Software Solutions, Inc.
727-437-BSS1 (727-437-2771)
@BSSI_Consulting

S52CQ, 2018-01-28:
Hi Tom,

I did set Radio One Keyer RTS: PTT to None instead of PTT. Then it just would not key the radio.

73s de Jure, S52CQ

S52CQ, 2018-04-18:
Hi Tom,

any new update regarding this issue?

73s de Jure, S52CQ

n4af, 2018-11-06:
closing per db4ple update:
please test the newest binary. It should fix the RTS problem with TR4W and UHSDR.
@ny4i: Mea culpa. It was our fault after all (in fact mine), everything in TR4W is working as it should. Please close the respective TR4W issue: n4af/TR4W#290 or have it closed.

73
Danilo

closed 2018-11-06 by n4af