TR4W/TR4W #305: CWByCAT SO2R failure to send exchange on second radio

n4af, 2018-01-18:
The following scenario under wk works:
1) CQ on R1 while S&P on R2
2) ALT-D station heard on R2
3) Break R1 CQ to call station on R2 via space bar
4) As soon as call dumped on R2, press Fkey to CQ on R1
   _** note: so far all works as expected WK vs CWBC_
5)Station called on R2 has given his exchange, NOW
5) Press ENTER this should log the R2 exchange AND give my exchange (Works using Wkey)
6) With CWBC, when #4 above is executed TR4W correctly logs the R2 QSO **BUT DOES NOT SEND EXCH**.
7) After logging R2 qso, the focus is entirely on R1, using CWBC the station called on R2 does NOT receive callers exchange.

n4af, 2018-01-18:
i have been able to simplify this. I suspect a timing problem.
Using two K3 radios, occasionally, I can get the failure by just 
1) ALT-D
2) space bar
3) press enter. Part of exchange (or none) will send before the active radio is switched back to run.
So the question is how CWBC detects end of alt-d sent exchange.

ny4i, 2018-01-19:
CWBC makes a very good guess on when sending is done. Is AddStringToBuffer involved in this? There is a comment about how CWBC figures out the timing or the message. If AddStringToBuffer is set, the Timer object may need to be reset. I admit I'm not as familiar with ALT-D. It is also possible that the timer from the previous send is terminated around the same time the new CW is desired and that stops sending.

closed 2018-11-15 by n4af