TR4W/TR4W #331: Support HolyLand contest

n4af, 2018-03-08:
UR7QM needs HolyLand QSO Point Method added to award double points on 160/80M.

ok1rr, 2018-03-08:
A friend of mine, an active user of TR4W reports that the speed changes 
within the macros are not functional since version 4.64. The old version 
4.244 works as expected.

CQ and CQ CW EXCHANGE lowers the speed from unknown reason.

He uses a config as follows. Does somebody have any recommendation for him?

73,
Martin, OK1RR
----------------------------------------------------

MY CALL=OK1VK
CATEGORY-ASSISTED=ASSISTED
CATEGORY-BAND=ALL
CATEGORY-MODE=CW
CATEGORY-OPERATOR=SINGLE-OP
CATEGORY-POWER=LOW
CATEGORY-TRANSMITTER=ONE
CONTEST=MINITEST

CALLSIGN UPDATE ENABLE=TRUE
DE ENABLE=FALSE
PREFIX MULTIPLIER=NONE
DUPE SHEET AUTO RESET=TRUE
WINDOW SIZE = 15
AUTO CALL TERMINATE = TRUE
AUTO SEND CHARACTER COUNT = 5
CW TONE = 0
WEIGHT = 1.06

[Commands]
LEADING ZEROS=3

[Messages]
CQ CW MEMORY F1= <06><06><06><06>TEST<13><13><13><13> \
CQ CW MEMORY F2=5NN #
CQ CW MEMORY F3=<13><13><13>#<06><06><06>
CQ CW MEMORY F4=<13><13><13>#-1<06><06><06>
CQ CW MEMORY F5=5NN
CQ CW MEMORY F6=DE OK1<13><13>VK<06><06>
CQ CW MEMORY F7=OK1<13><13>VK<06><06>
CQ CW MEMORY F8= NR?
CQ CW MEMORY F9= ?
CQ CW MEMORY F10= @
CQ CW MEMORY F11=5NN #-1
CQ CW MEMORY F12= :
EX CW MEMORY F1= DE OK1<13><13>VK<06><06>
EX CW MEMORY F2= 5NN #
EX CW MEMORY F3=<13><13><13>#<06><06><06>
EX CW MEMORY F4=<13><13><13>#-1<06><06><06>
EX CW MEMORY F5= 5NN
EX CW MEMORY F6=DE OK1<13><13>VK<06><06>
EX CW MEMORY F7=OK1<13><13>VK<06><06>
EX CW MEMORY F8= NR?
EX CW MEMORY F9= ?
EX CW MEMORY F10= @
EX CW MEMORY F11=<13><13>5NN #-1<06><06>
EX CW MEMORY F12= :

CQ CW EXCHANGE=^^~^<06><06><06><06><06><06>5NN 
<13><13><13><13><13><13><13><13>#<06><06>
CALL OK NOW MESSAGE => @_
CQ CW EXCHANGE NAME KNOWN=^^~^<06><06><06><06><06><06>5NN 
<13><13><13><13><13><13><13><13>#<06><06>
QSL CW MESSAGE= ^TU_\
QSO BEFORE CW MESSAGE= SRI QSO TU \
QUICK QSL MESSAGE= R^TU
REPEAT S&P CW EXCHANGE=^^~^<06><06><06><06><06><06>5NN 
<13><13><13><13><13><13><13><13>#<06><06>
S&P CW EXCHANGE=R^~^<06><06><06><06><06><06>5NN 
<13><13><13><13><13><13><13><13>#<06><06>

; SHORT 0 = 0
; SHORT 1 = 1
; SHORT 9 = 9
PADDLE MONITOR TONE = 0
; SAY HI ENABLE = TRUE
; SAY HI RATE CUTOFF = 70
; SHOW SEARCH AND POUNCE = TRUE

; SHORT 1 = 1
; SHORT 9 = N
; SHORT 0 = o
; SHORT INTEGERS = TRUE

n4af, 2018-03-08:
He cannot do it that way.
The speed UP/speed DOWN must balance to preserve original speed
CQ CW MEMORY F1= <06><06><06><06>TEST<13><13><13><13> \  *OK*

CQ CW EXCHANGE NAME KNOWN=^^~^<06><06><06><06><06><06>5NN  *NO/WRONG*