TR4W/TR4W #332: Add Icom IC-9700 support

ny4i, 2018-03-09:
Note that the current Icom enumeration goes through IC9100 so this will have to be changed to IC9700 everywhere. Or come up with a new enumeration upper limit (or use High(type)).

Obviously since this is a concept radio, this has to wait until we know its 2m/440 features.

ny4i, 2019-05-15:
I added this in a branch called 2019IcomFixes. Next in this build is to check PTT On by software and RIT.

The functions I tried seem to work on the radio.

I also recall someone had an issue with the 7610 but I do not have one to test. A friend has one so I might be able to check it there.

n4af, 2019-05-15:
Thanks Tom !

73, Howie

ny4i, 2019-05-15:
Don’t incorporate that branch just yet though.

I’m fixing a few more related Icom things.

Plus I want to check the 7300 too.

Thanks

Tom

Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

> I added this in a branch called 2019IcomFixes. Next in this build is to
> check PTT On by software and RIT.
>
> The functions I tried seem to work on the radio.
>
> I also recall someone had an issue with the 7610 but I do not have one to
> test. A friend has one so I might be able to check it there.
>
> <
https://github.com/n4af/TR4W/issues/332?email_source=notifications&email_token=ABVLCUBJ33BATASLXFKDBHTPVQMM5A5CNFSM4EUO7FV2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODVO2Q6Q#issuecomment-492677242
https://github.com/

ny4i, 2019-05-15:
Obviously my new shack and dev environment are back in action.

Tom

Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

> I added this in a branch called 2019IcomFixes. Next in this build is to
> check PTT On by software and RIT.
>
> The functions I tried seem to work on the radio.
>
> I also recall someone had an issue with the 7610 but I do not have one to
> test. A friend has one so I might be able to check it there.
>
> <
https://github.com/n4af/TR4W/issues/332?email_source=notifications&email_token=ABVLCUBJ33BATASLXFKDBHTPVQMM5A5CNFSM4EUO7FV2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODVO2Q6Q#issuecomment-492677242
https://github.com/

ny4i, 2020-08-05:
I verified the functions work for the IC9700.

closed 2020-08-05 by ny4i