TR4W/TR4W #337: Question on compatibility with winkey2 interfaces.

ur7hfo, 2018-03-15:
Question on compatibility with winkey2 interfaces.
I have a key from k3ng, it works with the program n1mm, and does not want to work with the program tr4w. Maybe you need additional settings from my side? Tell me if you can.

n4af, 2018-03-15:
I have no way to answer this.
TR4W supports the K1EL winkey-- I own 3 of them and all work as designed.

73, Howie N4AF

ok1rr, 2018-03-15:
Here works without any problem in both Windows and Linux. Works both 
Winkey and Tinykeyer (based on the K3NG Arduino keyer)...

73,
Martin, OK1RR

Dne 15.3.2018 v 10:18 ur7hfo napsal(a):
> Question on compatibility with winkey2 interfaces.
> I have a key from k3ng, it works with the program n1mm, and does not want to work with the program tr4w. Maybe you need additional settings from my side? Tell me if you can.
>

ur7hfo, 2018-03-15:
Sorry, I already figured it out myself. For work, it was necessary to disable the automatic reset at arduino. Disabling automatic reset, everything worked fine!

closed 2018-03-15 by ur7hfo