TR4W/TR4W #339: Print Contest Title not Contest Name in Cabrillo header

n4af, 2018-03-20:
When using custom configurations, the CONTEST name appears as the 'cutout' name - often General QSO.

Instead please use CONTEST NAME OR CONTEST TITLE so the user does not have to overwrite the generated cabrillo header, replacing General QSO or whatever generic name was used.

n4af, 2018-03-20:
4.69.6 changes PostUnit to print Contest Title/Name.

n4af, 2019-02-18:
This fix will need to be changed as it broke contest name in generated contest for contests like ARRL DX.

n4af, 2019-02-18:
This should be corrected in 4.78.3:

 T3 := ContestTypeSA[contest];        // 4.78.2
   if Contest = GENERALQSO then       // 4.78.3
    if length(ContestTitle) <> 0  then
     T3 := ContestTitle
      else
       if length(ContestName) <> 0  then      // 4.78.3
        T3 := ContestName;

ny4i, 2020-07-05:
IARU Cabrillo export shows correct as this:

```
START-OF-LOG: 3.0
CREATED-BY: TR4W v.4.89.6
CALLSIGN: W4GAR
CONTEST: IARU-HF
CLAIMED-SCORE: 50
CATEGORY-ASSISTED: NON-ASSISTED
CATEGORY-BAND: ALL
CATEGORY-MODE: CW
CATEGORY-OPERATOR: SINGLE-OP
CATEGORY-POWER: HIGH
OPERATORS: NY4I,W4CU,W4UB
CLUB: club
LOCATION: WCF
NAME: tom
QSO: 14000 PH 2020-07-03 0513 W4GAR         59  8      W1AW       59  ARRL         
QSO: 14000 PH 2020-07-03 0513 W4GAR         59  8      W3UY       59  8            
QSO: 14000 PH 2020-07-03 0513 W4GAR         59  8      4U1ITU     59  IARC         
QSO: 14000 PH 2020-07-03 0513 W4GAR         59  8      W4UB       59  8            
QSO: 14000 RY 2020-07-03 0633 W4GAR         59  8      DL9UIU     59  DARC         
QSO: 14000 DG 2020-07-03 0633 W4GAR         59  8      JA3ERD     59  45           
END-OF-LOG: 
```

This is the Cab file from an ARRL-DX:
```
START-OF-LOG: 3.0
CREATED-BY: TR4W v.4.89.6
CALLSIGN: W4GAF
CONTEST: ARRL-DX-SSB
CLAIMED-SCORE: 3
CATEGORY-ASSISTED: NON-ASSISTED
CATEGORY-BAND: ALL
CATEGORY-MODE: CW
CATEGORY-OPERATOR: SINGLE-OP
CATEGORY-POWER: HIGH
OPERATORS: W4GAF
CLUB: club
LOCATION: WCF
NAME: tom
QSO:  1800 CW 2020-07-05 0549 W4GAF         599 FL     I1IJK      599 K            
END-OF-LOG: 

```

It looks like it is working ( I tested this with my branch in case it does not work in the TRUNK).

It looks ok to me.

closed 2020-07-05 by ny4i