TR4W/TR4W #340: Add option to automatically download latest CTY.DAT

ny4i, 2018-03-22:
Add an option to retrieve the CTY.DAT file from this link: http://www.country-files.com/cty/cty.dat and place in the proper directory.

Extra credit to go check the file to see if there is a newer version. 

I recommend rather than using the Win32 API approach that the program currently does to make use of the Indy Library to retrieve a web page. This is with the TIdHTTP object.

n4af, 2018-03-22:
Sounds good Tom, but I do not how to do it with indy.

What I have done of late is just roll latest cty.dat into the monthly
update.

73, Howie

ny4i, 2018-03-22:
I can do it. I put an example of using TIdHTTP in the ticket in case anyone wanted to take a crack at it.

I put things in the issues so I do not forget them. 
Tom
Principal Solutions Architect, ITIL®v3 
Better Software Solutions, Inc.
727-437-BSS1 (727-437-2771)
@BSSI_Consulting

ny4i, 2018-03-22:
> it would be a nice add'n to the tools pulldown.
> There is api code to check/update tr4w that would apply to cty.dat but indy
> sounds like a great way to go.
>
> 73, Howie

Do you mean we already have code in there to download the CTY.DAT file? If so, I missed that.

Tom
Principal Solutions Architect, ITIL®v3 
Better Software Solutions, Inc.
727-437-BSS1 (727-437-2771)
@BSSI_Consulting

n4af, 2018-03-22:
> 727-437-BSS1 (727-437-2771 <(727)%20437-2771> <(727)%20437-2771>)

NO. Extant code commented out to check and update trlog - i think vc.pas
i just meant it could act as a template...

ny4i, 2018-03-22:
I understand. Better to do it with Indy so let me look at that.

Thanks,

Tom
Principal Solutions Architect, ITIL®v3 
Better Software Solutions, Inc.
727-437-BSS1 (727-437-2771)
@BSSI_Consulting

n4af, 2018-11-15:
add to 4.75.3 in absence of indy code change...

closed 2018-11-15 by n4af

ny4i, 2018-11-15:
Thanks for implementing this. Sorry I have been out of pocket for a long time. I do intend to get back to this stuff soon.

n4af, 2018-11-16:
FB Tom

73, Howie