TR4W/TR4W #349: + in CW message sends 000 instead of PREV #

ny4i, 2018-05-27:
The LIST of Commands states that + sends the Prev #. I read this to mean the previous QSO exchange NR. 

I needed this as I am operating remotely so I do not have a paddle to send the prior number. I must be missing something as after I entered the other stations NR, he asked for a fill on my number. I had already hit enter so if I hit the macro that has #, I get the next serial number.

n4af, 2018-05-27:
something like ctrl-j AUTO QSO NUMBER DECREMENT, but in pileups right now...

ny4i, 2018-05-27:
Thanks! That handles that but we should check if + is working as it is supposed to do. Good luck!

n4af, 2018-05-28:
yes, if i can remember, will remove += from the list of commands.
Lets keep this open and I will code against this issue #

n4af, 2018-05-28:
removed Prev # command list as this should not be done via command, rather by using CTRL-J AUTO QSO NUMBER DECREMENT = TRUE
changed in 4.71.5

closed 2018-05-28 by n4af

ny4i, 2018-05-28:
I’m curious if this was just a bug where the + calculation didn’t get set
properly. I see no harm to have both options for flexibility.

There might be a case where one wants the previous number when the callsign
field has something in it.

I have to check the code though.

Tom

Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771