TR4W/TR4W #363: TR4W & Kenwood TS-480

S52CQ, 2018-10-29:
Hi,

today I have tested TS-480 with original FTDI USB/RS232 adapter connected to DB9 connector on the radio and have found out that TR4W does not support it.

I have tested many possibilities in the Settings/CAT & CW Keying, without success. Frequency of the radio does not show on the screen.

It's interesting though that N1MM and HamRadio deluxe work without any problem with the same radio and USB/RS232 adapter.

Any idea what am I doing wrong?

73s de S52CQ, Jure

n4af, 2018-10-31:
Hello Jure -

We have many users using Kenwood & Elecraft with the exact same polling as
TS480.

--->>> Ensure your CTRL-J POLL RADIO is set to TRUE.

Ensure baud rate matches the radio and serial port is set correctly

I can send you a program to trace the response to polling, but am sure it
will show nil based on your report.

73, Howie N4AF

n4af, 2018-11-05:
closing: no response

closed 2018-11-05 by n4af