TR4W/TR4W #374: Access violation running TR4W using DXCluster

n4af, 2019-03-18:


n4af, 2019-03-18:
Highly intermittent problem may occur during a contest under unknown circumstances:

Faulting application name: tr4w.exe, version: 0.0.0.0, time stamp: 0x2a425e19
Faulting module name: tr4w.exe, version: 0.0.0.0, time stamp: 0x2a425e19
Exception code: 0xc0000005
Fault offset: 0x00001c8e
Faulting process id: 0x17f8
Faulting application start time: 0x01d4dcb23be5f33d
Faulting application path: C:\TR4W\tr4w.exe
Faulting module path: C:\TR4W\tr4w.exe
Report Id: 0f2e40ed-4c21-47f4-9e13-0a2f76af3ed0
Faulting package full name: 
Faulting package-relative application ID: 
- <Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
- <System>
  <Provider Name="Application Error" /> 
  <EventID Qualifiers="0">1000</EventID> 
  <Level>2</Level> 
  <Task>100</Task> 
  <Keywords>0x80000000000000</Keywords> 
  <TimeCreated SystemTime="2019-03-17T11:40:11.386956200Z" /> 
  <EventRecordID>18731</EventRecordID> 
  <Channel>Application</Channel> 
  <Computer>n4af</Computer> 
  <Security /> 
  </System>
- <EventData>
  <Data>tr4w.exe</Data> 
  <Data>0.0.0.0</Data> 
  <Data>2a425e19</Data> 
  <Data>tr4w.exe</Data> 
  <Data>0.0.0.0</Data> 
  <Data>2a425e19</Data> 
  <Data>c0000005</Data> 
  <Data>00001c8e</Data> 
  <Data>17f8</Data> 
  <Data>01d4dcb23be5f33d</Data> 
  <Data>C:\TR4W\tr4w.exe</Data> 
  <Data>C:\TR4W\tr4w.exe</Data> 
  <Data>0f2e40ed-4c21-47f4-9e13-0a2f76af3ed0</Data> 
  <Data /> 
  <Data /> 
  </EventData>
  </Event>

ny4i, 2020-07-05:
We might need to put a better fault dump handler in and extend the debug logging idea.

n4af, 2020-07-05:
I have had this happen, never been able to isolate...