TR4W/TR4W #379: Add UDP Server to receive messages from WSJT-X

ny4i, 2019-05-25:


ny4i, 2020-06-29:
This is needed more urgently. I'm not thrilled with the complexity of the current contest software integration with WSJT-X. I am thinking something that can co-exist with JTAlert to just receive the ADIF records of the contacts. It should also run standing with WSJT-X and do the callsign highlighting if a contact is a dupe or not.

The difference from N1MM for example is I do not need it to be the main driver. I do not need to start WSJT-X from N1MM nor use its own config file. That seems needlessly complex and confusing to the user.

Rig control is a factor and implementing the DX Commander TCP/IP interface does make sense but I am still thinking on it.

We do need to add more mode support than just DG and RY. We would need a mode of FT8 and FT4 *(but generate the ADIF properly with MODE MFSK and submode of FT4 but for FT8, the mode is FT8).

ny4i, 2020-06-30:
I saved the current UDP API into our [Wiki](url) for convenience. Please always check that it is the latest.

ny4i, 2020-06-30:
Great progress towards this change. I now understand the packet format sent by WSJT-X and most importantly, I have a UDP Server receive thread running in TR4W that displays a QuickDisplay message whenever I receive a message. I can log normally and the program displays the heartbeat was received from WSJT-X periodically. So next step is to make an object to parse out the messages to handle the logging. This means I have to add a better ADIF parser (this ties into another issue I entered). I also have to consider if we have to do any sort of locking of the log write as I would not want a UDPReceive to interrupt other logging functions (although it is doubtful both the UI and WSJT-X would be logging). I do have to figure out how I am sent the callsigns to decode. It seems that is not sent but perhaps I have to respond to a heartbeat to see that.

ny4i, 2020-06-30:
I found a [Delphi implementation](https://sourceforge.net/p/alarmejt/code/HEAD/tree/trunk/uudp.pas) of the WSJT-X UDP spec in SourceForge. What a great find and this makes things much easier. I do love open source!

n4af, 2020-06-30:
wow, great news.
open source rocks !

ny4i, 2020-07-01:
Using a little test app and some significant massaging of the data types in f5jmh's code (that is FreePascal which has better data types than Delphi 7), I was able to create some output from WSJT-X. There are still a few bugs such as frequency and time when a QSO is logged, but it is close. Here is a sample I dumped to a Memo.

> Decode: -1 0414 -23 0 951 ~ NX8E K0SWT +00 4:14:30 AM 0 15270000
> Decode: -1 0414 -10 3.408E-312 1787 ~ OE7UU AC2SB FN03 4:14:30 AM 3.40792534102725E-312 15270000
> Status: Frequency: 3505416960 Mode: FT8 DX Call: K4YT Report: -22 TX Mode: FT8 TX Enabled: 0 Transmitting: 0 Decoding: 0 RXDF: 1158 TXDF: 1158 DECall: NY4I DEGrid: EL87 DXGrid: FM19
> QSO logged: Date:01-Jul-2020 00:14:56 DX Call:K4YT DX Grid:FM19 Frequency:1458924288 Mode:FT8 Report sent: -21 Report received:-06 TX Power:100 Comments: Name:
> ADIF Record to log: 
> <adif_ver:5>3.1.0
> <programid:6>WSJT-X
> <EOH>
> <call:4>K4YT <gridsquare:4>FM19 <mode:3>FT8 <rst_sent:3>-21 <rst_rcvd:3>-06 <qso_date:8>20200701 <time_on:6>041454 <qso_date_off:8>20200701 <time_off:6>041454 <band:3>40m <freq:8>7.075158 <station_callsign:4>NY4I <my_gridsquare:4>EL87 <tx_pwr:3>100 <EOR>
> Status: Frequency: 3505416960 Mode: FT8 DX Call: K4YT Report: -22 TX Mode: FT8 TX Enabled: 0 Transmitting: 0 Decoding: -1 RXDF: 1158 TXDF: 1158 DECall: NY4I DEGrid: EL87 DXGrid: FM19
> Decode: -1 0414 -13 0 765 ~ K0SWT NX8E R-02 4:14:45 AM 0 15285000
> Decode: -1 0414 -9 1.362E-312 1584 ~ KK7SB W1NA RRR 4:14:45 AM 1.36232150022729E-312 15285000
> Decode: -1 0414 -18 0 952 ~ CQ AG5XI EL09 4:14:45 AM 0 15285000
> Decode: -1 0414 -14 3.408E-312 2002 ~ K0KP W8SSM EN76 4:14:45 AM 3.40792530118579E-312 15285000
> Decode: -1 0414 -9 3.408E-312 390 ~ KI4TYK F5RRS RR73 4:14:45 AM 3.40792530055339E-312 15285000
> Decode: -1 0414 -20 2.836E-319 2248 ~ CQ ZL3IO RF80 4:14:45 AM 2.83628265308084E-319 15285000
> Decode: -1 0414 -21 3.408E-312 768 ~ YO9HP WD5BYQ EM12 4:14:45 AM 3.40792528031646E-312 15285000
> Status: Frequency: 3505416960 Mode: FT8 DX Call: K4YT Report: -22 TX Mode: FT8 TX Enabled: 0 Transmitting: 0 Decoding: 0 RXDF: 1158 TXDF: 1158 DECall: NY4I DEGrid: EL87 DXGrid: FM19
> Status: Frequency: 3505416960 Mode: FT8 DX Call: K4YT Report: -22 TX Mode: FT8 TX Enabled: 0 Transmitting: 0 Decoding: -1 RXDF: 1158 TXDF: 1158 DECall: NY4I DEGrid: EL87 DXGrid: FM19
> Status: Frequency: 3505416960 Mode: FT8 DX Call: K4YT Report: -22 TX Mode: FT8 TX Enabled: 0 Transmitting: 0 Decoding: 0 RXDF: 1158 TXDF: 1158 DECall: NY4I DEGrid: EL87 DXGrid: FM19
> Status: Frequency: 3505416960 Mode: FT8 DX Call: K4YT Report: -22 TX Mode: FT8 TX Enabled: 0 Transmitting: 0 Decoding: -1 RXDF: 1158 TXDF: 1158 DECall: NY4I DEGrid: EL87 DXGrid: FM19
> Decode: -1 0414 -15 3.408E-312 1803 ~ W5BDB AB2EZ R+00 4:14:45 AM 3.40792528031646E-312 15285000
> Decode: -1 0414 -13 3.408E-312 1701 ~ CQ K3LA EN91 4:14:45 AM 3.40792528031646E-312 15285000
> Decode: -1 0414 -22 6.324E-322 725 ~ CQ VA2PM FN46                         a1 4:14:45 AM 6.32404026676796E-322 15285000
> Status: Frequency: 3505416960 Mode: FT8 DX Call: K4YT Report: -22 TX Mode: FT8 TX Enabled: 0 Transmitting: 0 Decoding: 0 RXDF: 1158 TXDF: 1158 DECall: NY4I DEGrid: EL87 DXGrid: FM19

ny4i, 2020-07-06:
I have TR4W logging contacts from WSJT-X. There is some cleanup to do to make sure the points are logged, it is places in band map and other things.

I have found this code is the classic definition of spaghetti code. While there is a single function called LogContact, it does not do everything. Then you have other functions like ParametersOkay that check some thing and call LogContact. So which is it.

A contact should be logged by populating the proper data structure ContestExchange and calling LogContact. Everything else should be handles in that function. AN even better way too do it would be to have a TContestLog class and it gets called with log.NewContact(ReceivedData). It can be given the station configuration and operator info as well as a link to radio info in the Create. This really needs a rewrite to simply this code. It functions great but it sure is hard to modify and has no concept of encapsulation.

n4af, 2020-07-06:
Sounds like you are making great progress Tom !
73, Howie

ny4i, 2020-07-11:
I added a tcp server to uWSJTX and lo and behold, it works. I wanted to make sure I did not interested with the threading of any other network interfaces. I added a very tailored server (hard coded messages) and it works!!! Now I have to add a bit more processing to tie the radio control in so for example, when WSJT-X wants to know the status of split mode, we inquire the radio control and return the actual state of split. But that is easy stuff (just parsing).

ny4i, 2020-07-15:
Howie - Can you grab this branch to your PC and build it again. I believe I made the pull request before the last set of changes were committed. I also forgot to disable the debug tracing. Thanks

n4af, 2020-07-15:
Done.
url via gmail.
73, Howie

n4af, 2020-07-15:
MINGW64 /d/newsrc/tr4w/src (iss423)
$ git pull origin WSJT-X-Implementation
From github.com:n4af/TR4W
 * branch            WSJT-X-Implementation -> FETCH_HEAD
Merge made by the 'recursive' strategy.
 tr4w/src/VC.pas | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

ny4i, 2021-05-02:
This issue was closed previously.

closed 2021-05-02 by ny4i