TR4W/TR4W #390: Enhancement - A Configurable Path for the Log File Directory

OH6QU, 2019-12-22:
It would be nice to have a default directory for TR4W log files as a configurable parameter in the .cfg-file. In case the directory mentioned in the cfg-file is not found then the program should revert back to the factory default directory.

Background for the request: I am saving my configuration & log files in the OneDrive and it is a pain in the ass to navigate there from TR4W default log directory.

73,
Jari OH6QU

n4af, 2019-12-22:
Hello Jari -

Try ctrl-J  BACKUP LOG FILE NAME

73, Howie

OH6QU, 2019-12-22:
Thanks for your comment Howie. However it does not solve the problem as when starting TR4W you first have to navigate a loooong path to OneDrive to open the cfg-file you need. I just realized that adding the configurable path to the cfg-file does not solve the problem. Instead it needs to be added as a "global" parameter while installing or configuring TR4W for usage. For example the same way as you configure radio/com/lpt-port once in TR4W, you specify to which directory to open/save cfg- and log-files.

73,
Jari OH6QU

n4af, 2019-12-22:
Have you mapped OneDrive to a network drive ??

73, Howie

OH6QU, 2019-12-22:
Yes - OneDrive is seen as a logical path C:\Users\XXX\OneDrive in Windows7. I have dedicated directory in OneDrive to store TR4W cfg and log files. By that way I can access them all at home and at my contest QTH.

OH6QU, 2019-12-22:
For example DX4WIN Logging Software remembers from which directory you opened and saved the log file last time. By that way there is no need for a new configuration parameter to set. I think this might be a good way to implement this feature into TR4W as well. E.g. if I have saved the TR4W log last time to the path C:\Users\XXX\OneDrive\TR4WLogs\2019 CQ-WW-CW OH6QU, the next time TR4W starts with path C:\Users\XXX\OneDrive\TR4WLogs as a default.

ny4i, 2019-12-22:
I put as shortcut to the DropBox (in my case) backup folder in the TR4W directory.

Tom
Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

OH6QU, 2019-12-22:
Using a shortcut leading to a log directory in OneDrive makes my life much easier until TR4W can memorize the location of cfg and log files as I described above - thanks for the hint Tom!

OH6QU, 2019-12-22:
A solution with shortcut seems to be working fine! As an administrator I created with the Command Prompt a soft (symbolic) link directory from TR4W to OneDrive by using a following command:  mklink /D "C:\TR4W\TR4W v4.84.0\OneDrive" C:\Users\jak\OneDrive\RA-toiminta\TR4W

Now whenever I select a OneDrive link directory under TR4W it automatically moves me into a TR4W log directory in the OneDrive.

73 & Thanks for the help,
Jari OH6QU

closed 2026-11-04 by ny4i

ny4i, 2026-11-13:
The new Browse button brings up a full Windows file chooser.

[image]

[image]

In the new program, the file is a single sqlite db file for the db and settings are json file--although I do not recommend editing that outside of the program, you could.