TR4W/TR4W #391: Two more telnet clusters to trcluster.dat

OH6QU, 2019-12-24:
The default list of telnet clusters (trcluster.dat) in TR4W lacks two important dx clusters that are widely used in the Northern Europe:

sk3w.se:8000
ham.connect.fi:7300

Please add them to the file.

n4af, 2019-12-24:
Hello Jan -

Could you build a list of valid Telnet sites ?
I am sure there are many on the list that no longer exist- I can get our members to supply you with suggestions ?
Once you have a good TRCLUSTER.DAT, I can integrate into the next build.

Tks, Howie

OH6QU, 2019-12-25:
Hi Howie,

Task accepted. Jim AD1C is already preparing for TR4W a new trcluster.dat file that contains almost 650 telnet sites. Jim is webmaster of dxcluster.info site. I contacted him and asked can we use the data found on his site. He agreed without hesitations :)

Jari

n4af, 2019-12-25:
FB Jan -

Tks, Howie

n4af, 2019-12-25:
Hi Jan -

See how this version of AD1C works ?

73, Howie

OH6QU, 2019-12-25:
I'll test it as soon as he sends the file to me.

Jari

OH6QU, 2019-12-25:
Ooops - I didn't look at my mailbox. I'll check is it working.

Jari

OH6QU, 2019-12-25:
Oops again - it was an old conversation in the mail box so no trcluster.dat file received yet. Jim said he will make after the holidays.

Jari

n4af, 2019-12-25:
hmmm, should be there but i also put on the tr4w web.
<http://tr4w.net/trcluster.dat>

OH6QU, 2019-12-25:
The file is not OK. Each telnet address is followed by information which cluster software is used (e.g. DX Spider). This information is not needed. Also each line is missing <CR> (ENTER).

Here is an extract how it should look like:
4m5dx.no-ip.org:7300
dxc.dx.or.kr:23
7n4twl.homeip.net:8000
dxc.ham.hr:7300
9a0dxc.hamradio.hr:8000

n4af, 2019-12-25:
> Also each line is missing (ENTER).

but when you select any entry and press enter it should connect ???

n4af, 2019-12-25:
You are selecting this from tr4w cluster window-- right ??

OH6QU, 2019-12-25:
I haven't tested the file with TR4W Cluster Window yet. I just made a visual inspection of the file content and I saw that the file content (or syntax) isn't the same as in the original file. I'll test it now if TR4W Cluster Window can interprete the file anyway.

OH6QU, 2019-12-25:
The file is working! I randomly tested around 20 telnet addresses by logging into them and most of them are working. I guess the ones that are not working are really offline at the moment because DX Cluster window didn't raise any error message, it just didn't able to reach the host etc.

However the pull down menu where you select telnet address to connect looks now a bit messy as the type of the cluster is now shown there together with address and port. Therefore my proposal is either to remove the information of cluster type from trcluster.dat-file OR add a new field in which the type of the selected dx cluster is shown. For example like this:
ham.connect.fi:7300             DX Spider
ham.crx.cloud:7300              OTHER
ham.homelinux.org:8000      DX Spider

Also my recommendation is to edit trcluster.dat in a way that there is a Carriage Return after each dx cluster. Otherwise the file is very difficult to read and edit manually.

ny4i, 2019-12-26:
I wrote a Perl script to do an NSLOOKUP on the file. I found 163 out of the file do not resolve to valid hostnames. 
Here is the file with valid hostnames or IP. Note I have not codes a check yet to see if I can connect to them. That would one next...

```
dxc.dx.or.kr:23,AR-Cluster
7n4twl.homeip.net:8000,DX Spider
9a0dxc.hamradio.hr:8000,DX Spider
9a0dxv.hamradio.hr:8000,DX Spider
62.150.41.82:8000,AR-Cluster
9m2pju.hamradio.my:7300,DX Spider
50.182.225.11:23,CC Cluster
dxspots.com:7300,CC Cluster
dxc.ai9t.com:7300,AR-Cluster
dxc.ai9t.com:7373,AR-Cluster
58.244.248.177:7373,AR-Cluster
spot.ba2ia.com:7373,AR-Cluster
lowband.club:7300,DX Spider
dx.5iyou.cn:7300,DX Spider
BG8FFE.XICP.NET:7373,DX Spider
BG8FFE.XICP.NET:7373,DX Spider
WWW.BG8NUD.COM:7373,AR-Cluster
103.207.69.56:7300,DX Spider
spots.bi4scc.net:7300,AR-Cluster
146.83.196.31:9000,DxNet
cluster.arlc.pt:7373,DX Spider
hamradio.isel.ipl.pt:41112,DX Spider
cluster.no-ip.org:7300,DX Spider
ct1enq.myvnc.com:35000,AR-Cluster
cu3hy.duckdns.org:7300,DX Spider
cx2sa.net:9000,DxNet
cx2sa.net:7300,DX Spider
www.db0erf.de:41113,DX Spider
db0fhn.efi.fh-nuernberg.de:7300,DX Spider
dx.prgm.org:41112,DX Spider
db0spc.ath.cx:8000,DX Spider
db0sue.de:8000,DX Spider
dd5xx.dxcluster.net:7000,DX Spider
rbn.dh8wr.com:7373,AR-Cluster
ccc.dh8wr.com:7373,CC Cluster
foc.dj1yfk.de:7300
dl0rtg-2.spdns.eu:7300,DX Spider
37.120.185.166:7373,AR-Cluster
dl8fcl.dyndns.org:8000,DX Spider
dl8las.dyndns.org:7301,CW Skimmer
dxc.funkerportal.de:443,OTHER
dxcluster.dq8z.de:23,AR-Cluster
e73du.no-ip.org:7300,DX Spider
185.73.168.155:23,AR-Cluster
campelo.cdf.udc.es:8000,DX Spider
ea2cw.ddns.net:7300,CC Cluster
rcfcloud.dyndns.org:7300,DX Spider
dx.ea2urv.com:7300,DX Spider
ea3aks.org:8000,DX Spider
ea3aks.org:41112,CLX
ea4fit.ddns.net:8000,DX Spider
ea4fit.ddns.net:7373,CC Cluster
ea4rch.dxfun.com:8000,DX Spider
eadx.org:7300,DX Spider
dx.ea5rci.com:8000,DX Spider
ea5urm.dns2go.com:7000,DX Spider
dxmaps.com:7300,DX Spider
ea7urg.es:8001,DX Spider
dx.ea7urm.org:8000,DX Spider
dx.ed1zac.net:8000,DX Spider
80.32.111.228:9000,DX Spider
mayodx.ath.cx:7300,DX Spider
shandx.ath.cx:5000,DX Spider
84.74.112.8:7300,DX Spider
212.24.100.199:7300,DX Spider
f1oyp-1.ampr.org:7000,DX Spider
[... 417 lines cut ...]
my $notFound = 0;
my $found = 0;

while (<>) {
   my $line = $_;
   my ($hostname,$rest) = split(/[:, ]/);
   if ($hostname =~ /\d{1,3}\.\d{1,3}\.\d{1,3}\.\d{1,3}/){
      print "$line";
      # IP address so skip the host check
      $found++;
   } else {
      my $nslookup = Net::DNS::Nslookup->get_ips($hostname);
      if (length($nslookup) == 0){
         #print "$hostname not found\n";
         $notFound++;
      } else {
         print $line;
         $found++;
      }
   } 
}

print "#Not found = $notFound\n";
print "#Found = $found\n";
```

ny4i, 2019-12-26:
And this is the valid list after I checked if the server responded to a telnet connect.
```
dxc.dx.or.kr:23,AR-Cluster
7n4twl.homeip.net:8000,DX Spider
9a0dxc.hamradio.hr:8000,DX Spider
9m2pju.hamradio.my:7300,DX Spider
dxspots.com:7300,CC Cluster
dxc.ai9t.com:7300,AR-Cluster
dxc.ai9t.com:7373,AR-Cluster
lowband.club:7300,DX Spider
dx.5iyou.cn:7300,DX Spider
WWW.BG8NUD.COM:7373,AR-Cluster
spots.bi4scc.net:7300,AR-Cluster
146.83.196.31:9000,DxNet
cluster.arlc.pt:7373,DX Spider
hamradio.isel.ipl.pt:41112,DX Spider
cx2sa.net:9000,DxNet
cx2sa.net:7300,DX Spider
www.db0erf.de:41113,DX Spider
dx.prgm.org:41112,DX Spider
db0spc.ath.cx:8000,DX Spider
db0sue.de:8000,DX Spider
dd5xx.dxcluster.net:7000,DX Spider
rbn.dh8wr.com:7373,AR-Cluster
ccc.dh8wr.com:7373,CC Cluster
foc.dj1yfk.de:7300
37.120.185.166:7373,AR-Cluster
dl8fcl.dyndns.org:8000,DX Spider
dl8las.dyndns.org:7301,CW Skimmer
dxc.funkerportal.de:443,OTHER
dxcluster.dq8z.de:23,AR-Cluster
185.73.168.155:23,AR-Cluster
ea2cw.ddns.net:7300,CC Cluster
rcfcloud.dyndns.org:7300,DX Spider
ea4fit.ddns.net:8000,DX Spider
ea4rch.dxfun.com:8000,DX Spider
eadx.org:7300,DX Spider
ea5urm.dns2go.com:7000,DX Spider
dxmaps.com:7300,DX Spider
ea7urg.es:8001,DX Spider
dx.ea7urm.org:8000,DX Spider
dx.ed1zac.net:8000,DX Spider
mayodx.ath.cx:7300,DX Spider
shandx.ath.cx:5000,DX Spider
84.74.112.8:7300,DX Spider
212.24.100.199:7300,DX Spider
f1oyp-1.ampr.org:7000,DX Spider
f1oyp.no-ip.org:7400,DX Spider
f1oyp.no-ip.org:7300,AR-Cluster
f1oyp.no-ip.org:7373,DX Spider
82.64.40.245:7300,DX Spider
ham.crx.cloud:7300,OTHER
dxcluster.f4gve.net:7300,DX Spider
lecluster.f5len.org:7373,DX Spider
f5mzn.org:9000,DX Spider
dxspider.sysinf.fr:7300,DX Spider
f6bvp.org:7300,DX Spider
clx.lunique.fr:9000,CLX
f6gcp-pat.dyndns.org:7300,DX Spider
dxcluster.arp75.org:7373,DX Spider
dxc.g3lrs.org.uk:7300,DX Spider
g6nhu.getmyip.com:7300,DX Spider
gb7baa.com:7300,DX Spider
gb7djk.dxcluster.net:7300,DX Spider
90.155.35.170:7300,DX Spider
GB7DXG.lebs.org.uk:7373,CC Cluster
gb7dxg-1.lebs.org.uk:7300,DX Spider
81.149.0.149:7300,CC Cluster
gb7cow.dyndns.org:9000,DxNet
www.gb7mbc.net:8000,DX Spider
[... 234 lines cut ...]
w9az.ddns.net:7373,AR-Cluster
134.48.91.82:23,AR-Cluster
dxc.w9pa.net:23,AR-Cluster
dxc.w9pa.net:7373,AR-Cluster
dxc.w9pa.net:7374,AR-Cluster
w9yb-dx.student-orgs.purdue.edu:23,AR-Cluster
w9yb-dx.student-orgs.purdue.edu:7373,AR-Cluster
dxc.wa9pie.net:8000,DX Spider
dxc.leadmon.net:7300,DX Spider
108.67.41.107:7373,CC Cluster
telnet.reversebeacon.net:7000,DX Spider
wb8zrl.no-ip.org:7300,DX Spider
dxc.wc2l.com:23,AR-Cluster
dxc.wc2l.com:7373,AR-Cluster
DXC.WD1L.net:7300,CC Cluster
dxc.wd4ixd.net:23,CC Cluster
ws7i.ewarg.org:7300,CC Cluster
wt3q.gofrc.org:7300,DX Spider
dxc.ww1r.com:7300,DX Spider
xq8fgc.cx2sa.net:9000,DxNet
yv6btf.no-ip.org:8000,AR-Cluster
olson.nz:9000,DX Spider
zl2arn.ddns.net:7300,DX Spider
zs2ez.dlinkddns.com:7373,CC Cluster
```

n4af, 2019-12-26:
Thanks Tom -

I will add to 4.85

73, Howie

OH6QU, 2019-12-27:
Thanks guys! Now TR4W will have an comprehensive list of DX clusters to select. And it will be easy to maintain up-to-date with the help of information & file available from dxcluster.info -site.

73,
Jari OH6QU

ny4i, 2019-12-27:
I used our list. I could scrape that site’s telnet list to build a comprehensive list and test them to produce a known working list. 

In addition to the ones we have of course. 

Tom

Principal Solutions Architect
Better Software Solutions, Inc. 
727-437-2771

OH6QU, 2019-12-27:
Tom,

To my understanding this list came originally from Tim AD1C/dxcluster.info and Howie just edited it to fit into TR4W syntax.

73,
Jari OH6QU

ny4i, 2019-12-27:
OK. Howie, If that sounds right, when was that last done?

It is easy enough to have a Perl script grab an HTML table to periodically grab a fresh list. I will check if they have a more programmer friendly way to like a file or JSON or XML.

Tom
Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

OH6QU, 2019-12-27:
Tom,

Tim AD1C has promised to produce TR4W compatible dx cluster file automatically from dxcluster.info. The file will be available here --> http://www.dxcluster.info/telnet/TRCLUSTER.DAT

It will be good to "clean" the TRCLUSTER.DAT file before taking it into TR4W release package by removing those dx clusters that are offline i.e. host is not reachable. Your code will do that automatically.

ny4i, 2019-12-27:
Perfect.

Once he realizes that, I will make sure I can automatically scan it.

One issue that would be useful is to allow comments in the files. The file is just sent to the TR utility function EnumerateLinesinFile (if I recall). I wonder if it would be useful to have an overloaded function to allow a comment to be ignored. I say that as I can hear the inevitable chorus now of “Why isn’t my clubs site in there” where we found it was removed.

Maybe I can accomplish that with a list on our website (or the Wiki on GitHub) with the comment on the ones I removed. Yes, maybe that is a better place for it.

Tom
Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

OH6QU, 2019-12-27:
In my opinion this scanning of telnet addresses should be done at dxcluster.info -site for example every 24 hours or so. It would be good to indicate for each telnet address when the host has been seen last time. If a host is unreachable certain time e.g. 10 days the address is marked as "no good" and it is no longer included in the list of active dx clusters. This could be proposed to Tim AD1C as a future development at his site.

Jari OH6QU

ny4i, 2024-02-06:
This has not been touched in a while but I believe it can be closed. Jim AD1C asked for my script to check the cluster info. I sent him that info and presume he uses them to check the if the clusters are active. MY scripts are on GitHUb here: https://github.com/ny4i/cluster

ny4i, 2024-02-06:
Well this did bring up a question for me. Howie - How do we get AD1C's updated DXCLUSTER file into TRCLUSTER when he updates it? Is that something you download periodically?

n4af, 2024-02-06:
> I'm thinking it could be added to the build script to download the file
> with cURL before the installer is built. But maybe you do that already.
>
> C:\Users\toms>curl -o TRCLUSTER.DAT http://www.dxcluster.info/telnet/TRCLUSTER.DAT
>   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
>                                  Dload  Upload   Total   Spent    Left  Speed
> 100 14837  100 14837    0     0  46722      0 --:--:-- --:--:-- --:--:-- 46804

Hi Tom -
Not sure i am subscribed to his dxcluster.
I think it would have to be manually added.
TRCLUSTER.DAT probably needs not only updates but delettions of inactive
clusters.

note: target/trcluster.dat changes ARE  picked up on compile

Tks, Howie

ny4i, 2024-02-06:
I see that the TRCLUSTER.DAT on the tr4w website has significantly less entries that the one on Jim's website. And when I do an install, the TRCLUSTER.DAT file in the install directory is from 2020 and it only has about 108 lines in it.

Did I understand you email Howie that the file is downloaded from Jim's site before each build? If so, something is happening with the file getting into the installer.

ny4i, 2024-02-06:
Oh wait. I always forget to look in target. One sec...