TR4W/TR4W #392: TR4W DX Cluster not automatically connecting to the server if it is not listed in trcluster.dat

OH6QU, 2019-12-24:
Following statement in .cfg-file does not open the connection nautomatically to the telnet server. In order to make it work you need to add the telnet server first to trcluster.dat-file.

;DX-cluster
;----------
TELNET SERVER = ham.connect.fi:7300
CONNECTION AT STARTUP = TRUE
CONNECTION COMMAND = OH6QU

ny4i, 2019-12-30:
I had a few minutes to look into this issue. It appears that the code in uTelnet.pas (line 345) does this:
```pascal
EnumerateLinesInFile('TRCLUSTER.DAT', EmunTRCLUSTERDAT, False);

i := SendDlgItemMessage(hwnddlg, 102, CB_FINDSTRING, -1, integer(@TelnetServer[1]));    //n4af 4.35.1
if i <> CB_ERR then tCB_SETCURSEL(hwnddlg, 102, i);
```
It appears that rather than set the value of the telnet server to whatever the user places in the config item for telnet server, it builds the drop-down from the file, then does a find.

It appears this should instead try the find and if that does not work, then set it to the actual value and try the connect. Maybe change to something like this:

```pascal
i := SendDlgItemMessage(hwnddlg, 102, CB_FINDSTRING, -1, integer(@TelnetServer[1]));    //n4af 4.35.1
if i <> CB_ERR then 
   begin
   tCB_SETCURSEL(hwnddlg, 102, i);
   end
else
   begin
   //pseudo code
   // Add an entry to the combo box of the value in the config
   // Set the index, I, to the new value?
   end;
```

n4af, 2019-12-30:
Thanks Tom -

I am leaving for an overnight in Charlottesville Va and will have a look on
return.

If you have a chance, the ctrl-end function to switch focus to the bandmap
does not work when telnet active ?
I would like to see it put the cursor on first entry in the b.m.

Tks, Howie

ny4i, 2020-07-01:
This has been fixed and is awaiting integration into the main code stream.

ny4i, 2020-07-03:
This did not make it into the code for some reason. I have to check my local branch and compare to master.

ny4i, 2020-07-22:
Confirmed as fixed in August 2020 release.

PR #412 (2020-07-02): Telnet connect not in list
The fixes #392 .  Now an item in the cluster config is added to the server drop-down list at run time but does not modify TRCLUSTER file. Also added better messages if the the cluster cannot connect so this also fixes #67 .
commit 2c4e6da3 (2020-07-02): Alow connecting to cluster not in file

This allows a user to connect to a cluster in the telnet config option even if the cluster is not in the TRCLUSTER file. This closes #392. Also closes #67 to give a better error message if we cannot connect.
closed 2020-07-22 by ny4i