TR4W/TR4W #394: Add support for the Ten Ten Omni VII

ny4i, 2020-01-18:
Parke Allen VY1JA ( vy1ja@northwestel.net ) asked if we could support the Omni VII.

I found the manual here: [https://www.tentec.com/wp-content/uploads/2016/05/tt_588_program_ref_v1.009.pdf]

The commands look similar but I did see some differences from the Omni code in the program.

n4af, 2020-01-18:
FB Tom -

Hope you are able to help. I guess a while until testing -- unless, someone
on the forum has an O7.

Tks, Howie

ny4i, 2020-01-18:
I agree. But that’s why I asked him when the rig was back and if he was ok testing new EXE files I can send.

Tom
Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

ny4i, 2020-01-18:
I see the Omni VII is supported in Hamlib. One more reason I should just add support for Hamlib and we get the benefit of all that work.

https://github.com/Hamlib/Hamlib/wiki/Supported-Radios

Tom
Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

n4af, 2020-01-18:
yes that would be optimal. HamLib would be in C tho.

73, Howie

ny4i, 2020-01-18:
They have a network daemon that could handle that but IO also saw bindings to allow Delphi to link to it.

There is also Omnirig which is in Delphi. http://www.dxatlas.com/OmniRig/ Since it supports the Omni 7, that might be a good candidate too. It doesn’t;t appear to support as much as Hamlib though.

Whatever I used, needs to support CW sending and RIT (Omnirig doesn’t).

For the Omni 7, I think the direct approach is fine as the commands are really close to the Orion 2.

Tom
Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

n4af, 2020-07-04:
VY1JA is QRT. Not sure if we have any other Omni VII users ?

ny4i, 2020-07-04:
QRT as in no longer operating? I recall he said the radio was being repaired. 

If he was only one, then I agree but worth the time. 

Tom

Principal Solutions Architect
Better Software Solutions, Inc. 
727-437-2771

n4af, 2020-07-04:
no. permanent- alzheimers.

73, Howie

closed 2020-07-04 by ny4i