TR4W/TR4W #397: Add new ARRL section of PE

ny4i, 2020-04-03:
Effective April 1, 2020, there is a new section abbreviated PE for Prince Edward Island. Would that just one a change to ARRLSECT.DOM?

ny4i, 2020-06-02:
I checked the June release and while PE is in the list of sections, the program does not let you enter it. You have to enter PEI. That is because the DOM file has PE = PEI. Shouldn't that be PE = PE or PE = PE, PEI?

n4af, 2020-06-02:
Good find Tom- btw, thanks the mult in WPX ;-)

Could you verify this fixes it <https://tr4w.net/4.89/tr4w_setup_4_89.1.exe>
?

73, Howie

ny4i, 2020-06-02:
Yes, that fixes it. Thanks.

I figured it is hard for the NY4 calls to find another :) I did see NY4F on as well.

73,

Tom 
Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

> On Jun 2, 2020, at 7:16 AM, Howie Hoyt <notifications@github.com> wrote:
> 
> https://tr4w.net/4.89/tr4w_setup_4_89.1.exe <https://tr4w.net/4.89/tr4w_setup_4_89.1.exe>

closed 2020-06-28 by n4af