TR4W/TR4W #399: Request for Support for UKSMG Summer Contest

G8FXM, 2020-04-30:
Could we add The UKSMG Summer ES contest as a supported contest please.
This is a 50Mhz contest which is very similar to the EUROPEAN VHF supported contest. It differs in that membership numbers are optionally exchanged in the contest QSO, like TEN TEN numbers. However these membership numbers need to appear in the Cabrillio and EDI exports and form part of the scoring.

Contest Rules can be found here: http://uksmg.org/summer-contest-rules.php
EDI File format can be found here: http://uksmg.org/contest/edi-file-format.php
Cabrillo File format here: http://uksmg.org/contest/cabrillo-file-format.php

EXCHANGE RECEIVED = RST QSO NUMBER GRID SQUARE AND POSSIBLE MEMBERSHIP NUMBER
OR reuse existing TEN TEN field as follows:
EXCHANGE RECEIVED = RST QSO NUMBER GRID SQUARE AND POSSIBLE TEN TEN

Scoring as per above rules website.

G8FXM, 2020-04-30:
Forgot to add, ADIF export to include UKSMG Membership number and contest name as defined here:
http://adif.org.uk/310/ADIF_310.htm#QSO_Field_UKSMG
Contest ID Enumeration = UKSMG-SUMMER-ES

n4af, 2020-04-30:
Could you advise how many UKMSG users would submit with TR4W  ?
This helps prioritize the number of hours required to modify TR4W-
alternatively,
since TR4W is open source, anyone is welcome to make the changes.

73, Howie N4AF

G8FXM, 2020-04-30:
Difficult to say, currently they are limited to choosing N1MM only as the others e.g. TacLog etc are old DOS products that have not been maintained and now no longer install. As TR4W runs on both Windows and Linux, we will promote this as the preferred logger for our test. Contest takes place annually expect circa 100 or so entrants, so not big in the grand scheme of things. Not expecting a high priority on this, so when ever you can fit it in.

G8FXM, 2020-05-04:
Not too concerned about scoring initially, my first objective is to get TR4W to accept the QSO exchange, i,e, RST, QSO, GRID, Possible USR1 or TEN TEN for membership numbers. I can manage with exporting ADIF and creating mylog file externally.

n4af, 2020-05-04:
you can try GENERAL QSO.

73, Howie

G8FXM, 2020-05-04:
I Tried that but membership numbers entered in exchange window are lost.

n4af, 2020-05-05:
ok, try this:
1) Install 4.88.3 <https://tr4w.net/4.88/tr4w_setup_4_88.3.exe>
2) Open up General QSO
3) Under FILE choose Open Log Directory
4) One of the files will be GENERAL QSO.CFG, EDIT IT.
5) Add MY STATE, MY NAME, and EXCHANGE RECEIVED as below, save the file,
exit and restart TR4W
*my state = 1702*
*my name = fm15mi*
*exchange received = QSO NUMBER NAME DOMESTIC OR DX QTH*
6) Lastly, CTRL-J * change CALLSIGN UPDATE ENABLE TO FALSE*
----------
*where my state is uksmg member #*
*my name is grid loc*
*--note: NOT case sensitive*

*EXAMPLE:*

*;Created by TR4W v.4.88.2[COMMANDS]MY CALL=N4AFmy state = 1702my name =
fm15miexchange received = QSO NUMBER NAME DOMESTIC OR DX
QTHCATEGORY-ASSISTED=NON-ASSISTEDCATEGORY-BAND=ALLCATEGORY-MODE=CWCATEGORY-OPERATOR=SINGLE-OPCATEGORY-POWER=HIGHCATEGORY-TRANSMITTER=ONECONTEST=GENERAL
QSO*

73, Howie

G8FXM, 2020-05-07:
Hi Howie,
This is great, thank you very much, I can use this in the upcoming TEST, My observations are I have to enter the qso exchange in this order, RST, SRX, GRID, MEMBERSHIP# any other order and I see membership# being recorded as SRX and visa versa, but I can live with that, I need to adjust the layout of my INITIAL.EX file now. Also EDI export isn't quite right, but Cabrillio if close to perfect and again I can live with that. Many thanks for finding the time for this. Best 73's, Dai, G8FXM

G8FXM, 2020-06-03:
A little feedback, as fyi in case needed in future, using the above config; 

1. EDI exports only export 4 digit grid squares, we would need 6 digit grids squares.

2. PExch= should be added to station info when creating an EDI file and ideally be pre-populated with 'MY_STATE' as this contains the stations membership number, however this can easily be edited afterwards so no big deal.

closed 2020-06-03 by G8FXM