TR4W/TR4W #400: 7QP gives "Improper domestic QTH!!" when logging a state in V4.88

FastThenLeft, 2020-05-02:
V4.88, Win10. First time install from scratch. I picked "7QP" for the contest. I'm in Indiana, so I put "IN" in the state.

I can log any of the 7QP counties like "AZNVO" or "WYCAM" but when I try and log a normal state like "TN" or "KY" I get "Improper domestic QTH!!" and it won't let me log the QSO.

n4af, 2020-05-02:
Those other states, PLEASE provide details of *WHICH* contest allows
logging of those other states - like TN or KY ?

What is your call ? I do not know what 'fasttthenleft' means ?

73,  Howie N4AF

FastThenLeft, 2020-05-02:
Sorry Howie. I usually do a much better job, but it was 6am after 7 hours
of debugging station items. I'm sure it's operator error, but after two
more hours I haven't figured it out. Here's further info:

OS: Win10
TR4W: 4.88 first time install.
Contest: 7QP - http://ws7n.net/7QP/new/Page.asp?content=rules
"MY CALL=N9KT"
"MY STATE=IN"

Exchange: 7th area stations send signal report plus 5-letter state/county
code (e.g., ORDES; see list).    County-line stations send multiple codes,
e.g., UTRIC/IDBEA (state code needed only once, e.g., ORDES/JEF).
 Non-7th-area stations send signal report plus state/province/"DX"
two-letter codes.    Stations in other QSO parties send their appropriate
exchange.    The 13 "Provinces" are VE1-9, VO and VY0-2.    County-line
contacts may be logged with one entry showing all counties or with separate
entries for each county.

Scoring: 2 points per SSB QSO, 3 points per CW QSO, 4 points per Digital
QSO.    County-line contacts count as multiple QSOs for both stations.
 7th area stations multiply total QSO points by the total of states (50),
provinces (13) and other DXCC entities (maximum of 10).    Non-7th-area
stations multiply total QSO points by 7th area counties worked (259).

Based on the above, I should work everyone and working someone in the 7th
callsign area counts as a multiplier.

What happens is this:  If I work a 7th area callsign, everything works
fine. For example:  Call "W7QP" with the exchange "WYCRO" works fine. But,
if I work call "W6TT" with the exchange "CA", the program won't allow me to
log it and flashes "Improper domestic QTH!!"

Just for comparison I tried two other logging programs and they allowed me
to log everyone and the scoring works fine, but I would really rather be
using TR4W!

What am I doing wrong? Thank you so much for your tireless hours writing
the program and helping idiots like me.

Thanks!
-David, N9KT

n4af, 2020-05-02:
OK, here are the applicable rules for 7QP:

1300 UTC Saturday to 0700 UTC Sunday (6 AM to midnight PDT the first
Saturday in May).   * 7th call area stations work everyone, others work 7th
area stations only.**..."*

TR4W will only log valid qsos--

This translates to Indiana can only work 7th call area.

AFAIK,  ALL state qso parties work this way ?

73, Howie

ny4i, 2020-05-02:
Are you operating from within a 7-land state? I ask as I am curious why you would log TN or KY. And if you are in the 7th call area, are you able to work other 7th call area stations according to the rules?

What is your callsign please?

Tom NY4I

Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

FastThenLeft, 2020-05-03:
Howie-

I know it says that, but that's not quite right. Let me explain.

If you are *running* stations, you *can't guarantee that everyone you work
will be a 7th area callsign*. If you are NOT a 7th area callsign, the
logging program should log everything but only score the 7th area contacts
- that's what N1MM+ does when you choose the 7QP contest and you are not in
the 7 area. That way I can log any station that comes to me while I'm
running. Otherwise I would be stuck doing S&P. From having worked this
contest for years, I know that's how it works. When you send in your log,
they strip out non-7 callsigns and then score the log.

Also, I am the president of the club that runs the INQP on the same day and
that's what we do too. You send in your log and if you are outside Indiana,
we strip out non-Indiana contacts and then score the log.

The DEQP and NEQP also run at the same time and that's how they work too.

What I ended up doing is selecting the 7QP in N1MM+ and it let me log
everything while highlighting and scoring the 7th area counties. When the
contests ended, I just send the Cabrillo to all four QSO parties and they
strip out the appropriate contacts and score the file.

Should I just have been running TR4W in a general QSO mode?

n4af, 2020-05-03:
Hi David -

> Howie-
>
> I know it says that, but that's not quite right. Let me explain.
>
> If you are *running* stations, you *can't guarantee that everyone you work
> will be a 7th area callsign*. If you are NOT a [...]

Yes,  since TR4W cannot properly score outside of declared contest (mults
or points), I would recommend
using General QSO- which will use WYSIWYG to handle multiple contests,
which the sponsor can then process.

You would need to ctl-J MY NAME to represent the sent exchange, else you
can run text processing against the cabrillo to change sent exchange
globally.
Also you would need to manually set F keys for various exchanges..

73, Howie

FastThenLeft, 2020-05-03:
Thank you Howie! I'll also make sure to write up some instructions for
using TR4W and include them on the INQP software page:
http://www.hdxcc.org/inqp/software.html
-David, N9KT

n4af, 2020-05-03:
Hi David -

Do I need to add INQP to the directly supported contests on TR4W ?

73, Howie

n4af, 2020-05-03:
David -

From TR4W forum:

Some weekends, like this one, feature multiple state qso parties.

To exploit this situation, consider adding a line to TR4W .CFG file. From within TR4W- set to your PRIMARY qso party.
OPEN LOG DIRECTORY and edit the .CFG file.

ADD a line at the bottom reading
DOMESTIC MULTIPLIER = WYSIWYG
Then restart TR4W

now you can do multiple contests from TR4W

The downside is you (may) need to edit the cabrillo sent exchange.and the score will no longer calculate correctly

But, not a problem for the sponsor as they will null  domestic mults not in their germane section.and calculate the points for you.

BTW, 4.88.2 will add the Indiana Q.P.

73, Howie

FastThenLeft, 2020-05-03:
Howie-

Awesome! Thank you! Here's some links to make it easier:
Home page: http://www.hdxcc.org/inqp/index.html
Rules: http://www.hdxcc.org/inqp/rules.html
County list: http://www.hdxcc.org/inqp/counties.html

-David, N9KT

n4af, 2020-05-04:
Hi David -

Could you test it out <https://tr4w.net/4.88/tr4w_setup_4_88.2.exe> ?
Also ensure the cabrillo is ok ?

I am also interested in the multi q.p. tool
<http://www.b4h.net/cabforms/multiqplogparser.php>

73, Howie

FastThenLeft, 2020-05-04:
Give me till Thursday morning and I'll give it a through test. Thanks!

n4af, 2020-06-28:
no feedback.
closing

closed 2020-06-28 by n4af