TR4W/TR4W #403: ARRL Field Day contest ID is wrong in ADIF export. Change this to ARRL-FIELD-DAY

ny4i, 2020-06-28:
The names exported in an ADIF file for the contest ID in each QSO record should match the ADIF 3 specification: http://adif.org.uk/310/ADIF_310.htm

Field-Day for example is exported as <CONTEST_ID:7>ARRL-FD when it should be <CONTEST_ID:14>ARRL-FIELD-DAY 

Even if the internal name needs to remain ARRL-FD, the exported name should be change to the proper value. The proper ADIF contest name could be added to the contest data and change upon export.

n4af, 2020-06-28:
4.89.6 <https://tr4w.net/4.89/tr4w_setup_4_89.6.exe> should change the
contest name to arrl-field-day.

you need to export your adif , start a new instance of arrl-field-day and
import the adif from arrl-fd into arrl-field-day..

73, Howie

n4af, 2020-06-29:
fixed 4.89.6

closed 2020-06-29 by n4af

ny4i, 2020-06-29:
Howie,

I should have grabbed this bug before you started. I have been working on this today to fix that but to also add the missing SRX_STRING and TRX_STRING. I will make that a separate Issue and use that for my fix. The program did not include the contest exchange in the ADIF record either.

Let me see your changes and how they merge with mine.

ny4i, 2020-06-29:
It doesn't look like you checked that into MASTER yet. Can you just not add that branch and my change can do it? That would eliminate any messy merging.

n4af, 2020-06-29:
sure.

73, Howie