TR4W/TR4W #415: Translation updates 2020-July

ny4i, 2020-07-03:
- [ ] WagWarn =     'Warning: Out of WAG allowed frequency range'; TC_WAGWARN

n4af, 2020-07-03:
Fixed 4.31.4

ny4i, 2020-07-03:
I'm confused then. I happen to see it int he code and opened this issue as a way to keep track of them. I searched in the language file but do not see them and I see that literal string in the program.

I figured we could keep this open until all I add are converted.

n4af, 2020-07-03:
Hi Tom -

We can certainly do that.
Right now it just goes out in english.

73, Howie

73, Howie

ok1rr, 2020-07-04:
Czech:

- [ ] WagWarn =     'Pozor: Mimo povolený rozsah frekvencí WAG'; TC_WAGWARN

73,
Martin, OK1RR

Dne 04. 07. 20 v 0:08 Tom Schaefer napsal(a):
> - [ ] WagWarn =     'Warning: Out of WAG allowed frequency range'; TC_WAGWARN
>

n4af, 2020-07-04:
Thanks Martin -

I will add.

73, Howie

n4af, 2020-07-04:
english and czech are are available @( https://tr4w.net/4.90/tr4w_setup_4_90.3.exe ).

n4af, 2020-07-04:
i have pushed branch wagwarn to repo but it is not merged into local or remote- since the xlations are incomplete.

ny4i, 2020-07-04:
I should have elaborated on my thoughts for this ticket. As I work through any issues or even create new messages, I figured I would collect all the strings that needed to be translated and we could do them at once. 

Tom

Principal Solutions Architect
Better Software Solutions, Inc. 
727-437-2771

ny4i, 2020-08-03:
- [ ] TC_WANTTOCONVERTLOG                   = 'Would you like to convert this log to the latest format?';
- [ ]   TC_LOGFILENOTFOUND                    = 'Log file not found';
- [ ]   TC_CANNOTBACKUPLOG                    = 'Could not make a backup copy of ';
- [ ]   TC_CANNOTCOPYLOGREADONLY              = 'Cannot copy log file -- target exists and is read-only';
- [ ]   TC_BACKUPCREATED                      = 'Log file backup created';
- [ ]   TC_CANNOTRENAME                       = 'Can not rename';

ok1rr, 2020-08-04:
Czech translations:

- [ ] TC_WANTTOCONVERTLOG                   = 'Chcete tento log 
konvertovat na nejnovější formát?';
- [ ]   TC_LOGFILENOTFOUND                    = 'Soubor logu nenalezen';
- [ ]   TC_CANNOTBACKUPLOG                    = 'Nelze vytvořit zálohu 
souboru ';
- [ ]   TC_CANNOTCOPYLOGREADONLY              = 'Nelze zkopírovat log 
-- cílový soubor již existuje a je pouze ke čtení';
- [ ]   TC_BACKUPCREATED                      = 'Záloha logu vytvořena';
- [ ]   TC_CANNOTRENAME                       = 'Nelze přejmenovat';

73,
Martin, OK1RR

Dne 03. 08. 20 v 22:11 Tom Schaefer napsal(a):
> - [ ] TC_WANTTOCONVERTLOG                   = 'Would you like to convert this log to the latest format?';
> - [ ]   TC_LOGFILENOTFOUND                    = 'Log file not found';
> - [ ] [...]