TR4W/TR4W #439: Allow user to set multiplier width

n4af, 2020-08-04:
Allow dynamic stipulation via CTRL-J of MuliplierWidth as is done for BandMapWidth.
The width of the multiplier significantly varies by contest.

Add new field to CommandsArray  - Multiplier Item Width that dynamically changes
BaseColumnWidth in VC.pas

ny4i, 2020-08-04:
I modified the commands too in my branch so we will have some conflicts. I’ll check in tonight. 

If you check this in, I can update mine to merge my changes in to your VC changes. 

Tom

Principal Solutions Architect
Better Software Solutions, Inc. 
727-437-2771

n4af, 2020-08-04:
No problems Tom.
I told Valery that I d/n plan to code it until after 4.91.
So you can hijack the issue or close it in reference to any extant one you
have.

Thanks, Howie

ny4i, 2020-08-05:
Are you thinking you would vary this in the FContest new contest settings or just let the user have a per-contest setting?

That also has me thinking that a way to read <contest>.cfg might not be a bad idea. We could apply it at the end of reading all other CFG files. so if WWDIGI.CFG exists, we automatically load it. It would go by the internal contest name we use so WWDIGI.CFG or ARRL-FD.CFG.

Unless we have another way to do that.

n4af, 2020-08-05:
> That also has me thinking that a way to read .cfg might not be a bad idea.

Hi Tom -

The latest pull went into 4.90.16
<https://tr4w.net/4.90/tr4w_setup_4_90.16.exe>. Perhaps a good point to
roll all into 4.91.0 ?
I was thinking of offering Multiplier Columnn Width as a a CTRL-J option.
Reading .cfg sounds good to me.

Thanks, Howie

ny4i, 2020-08-05:
OK. I was surprised 4.91 was not on the website yet. I may have misunderstood when you were releasing it.

I would have put what we had before these changes as it and save this for later but we can ship everything in MASTER now.

Let me know when you plan to put it on the website so I can refer people there this weekend.

Thanks,

Tom
Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

n4af, 2020-08-05:
ok. a communications failure as i did not think you were qrv.

Here is 4.91.0 <https://tr4w.net/4.91/tr4w_setup_4_91.0.exe>

73, Howie

n4af, 2020-08-10:
This was handled by option of window MULTS type:
WINDOW/MULTS selection.

If DEFAULT is chosen the mult window width is 5 characters wide.
If 'type' of mult (Domestic, Prefix, DX) is selected the window width is 13 characters wide,

Changed in 4.91.4

closed 2020-08-10 by n4af

n4af, 2020-08-10:
note: the title ctrl-j REMAINING MULT DISPLAY MODE is germane to a situation where there is a file containing known mults.
Where such does not exist, the HILITE option will display worked multipliers.
ARRL SWEEPSTAKES matches the first example and 
RDAC contest matches the second statement.