TR4W/TR4W #445: Add RTTY to ALT-M mode selection

n4af, 2020-08-18:
When selecting a mode with Alt_M RTTY is not there.
  There is only CW, SSB and DIGI. I had to choose DIGI and work.
  There were no comments during RTTY QSOs either.
  Everything worked as in the older versions.
  When forming the cabrillo, the RTTY mode was chosen (see clip).
  The heading of the log reflected:
  CATEGORY-MODE: RTTY
  but in the lines of QSO: it was written as DG.
  I have to edit the log from DG to RY manually.
  I think we need to add RTTY mode to Alt_M. If possible...

n4af, 2020-08-18:
4.91.8 reserved for this.

ny4i, 2020-08-18:
Howie,

There should not be a RTTY there. The modes int he program are CW< Phone, Digital and FM. The RTTY is derived in the Cab file for example based on the contest and the extendedMode. 

The issue where it showed mode DG int he Cabrillo file was a bug in the CAB generation and does not warrant displaying something else in the mode.

Tom
Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

n4af, 2020-08-18:
ok. well, i guess i can fix the cabrillo generation then.

tks, Howie

ny4i, 2020-08-18:
I thought I fixed that already. Perhaps not.

Tom
Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

n4af, 2020-08-18:
dunno. do see on tGenerateLogPortion in PostUnit:

if TempRXData.Mode = Digital then
       begin
       if TempRXData.ExtMode = eRTTY then
          begin
          ModeString := 'RY';
          end
       else
          begin
          ModeString := 'DG';

ny4i, 2020-08-18:
I would change that to the following:

if TempRXData.Mode = Digital then
   begin
   Case TempRXData.ExtMode of 
      eRTTY: ModeString := 'RY’;
      eFT8, eFT4: ModeString  ‘DG’;
      Else
         ModeString := ‘RY’;
      End;
   End;

Leave the two branches for clarity.

Tom
Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

n4af, 2020-08-18:
ok, i will forward that change to Val for testing.

tks, howie

n4af, 2020-08-19:
closing. rtty will be under the umbrella of DIGITAL but will generate RY / RTTY in cbr/adif.

closed 2020-08-19 by n4af