TR4W/TR4W #447: Prevent second radio bandmap qsy to band first radio is on

n4af, 2020-08-31:
An excellent way to blow a rx front end.

It is possible to click on a spot, with QSY INACTIVE RADIO checked- where the spot is on the same band the run radio is on. 

Check and disallow two radios on the same band.

n4af, 2020-09-04:
In searching on an algorithm to accomplish this, I am looking at the following.
given:
RADIO ONE = Band A
RADIO TWO = BAND B

QSY SPOT = BAND C
if bandC = bandA then QSY RADIO ONE
if bandC = bandB then QSY RADIO TWO
if bandC <> BandA or BandB then
normal rules apply (for now)
---------------
So, if the spot reflects a band that one of the two radios is on, QSY THAT RADIO (in band).

n4af, 2020-09-04:
not quite that simple-->
QSY RADIOx -only if NOT sending-, else exit.

ny4i, 2020-09-04:
Is that made simpler by using two band maps?

Screens are bigger nowadays. 

Tom

Principal Solutions Architect
Better Software Solutions, Inc. 
727-437-2771

n4af, 2020-09-04:
RE: removing chance of second radio on same band as first radio, I am not
sure 2 bm is germane.

Complexity certainly is amplified- folks would possibly want the option of
the second b.m. either tracking other radio OR
Band Map All Bands.

I am not sure TR4W paradigm of ONE entry window would work with confusion
of TWO b.m.?
To my way of thinking, you would need two entry windows for two b.m.- a-la
N1MM.

Since so2r is like athletics (no time to THINK how to catch or hit a ball),
things have to be intuitive.
I well remember N4OGW (now owner SO2SDR)  complaint re: 2 entry window
confusion.

For me, TR4W BM needs to map the S&P radio -or, all bands.

ny4i, 2020-09-04:
I have no idea what I am talking about as I am not a SO2R op so take this all with the proper amount of NaCl.

Can we not just set a rule in SetFreq (in the RadioObject) that it will not change if the other radio is on the same band (in two radio mode).

Tom
Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

n4af, 2020-09-04:
good question Tom.
it depends on how hard it is to get the SpotBand, ActiveBand, and
InactiveBand.

> I have no idea what I am talking about as I am not a SO2R op so take this
> all with the proper amount of NaCl.
>
> Can we not just set a rule in SetFreq (in the RadioObject) that it will
> not change if the other radio is on the same band (in two radio mode).
I think #1244 of LogRadio is in charge of that. But he seems to only know
Radio1/Radio2 ?
 I was looking at an earlier point in the code, perhaps
uBandMap TuneRadioToSpot ?
if you can get the SpotBand (EntryBand to uBandMap),
InactiveRadioPtr.BandMemory and ActiveRadioPtr.BandMemory
we should be ok.

Tks, Howie

ny4i, 2020-09-05:
Pseudo code…

If TwoRadioMode then
   begin
   If InActiveRadioPtr^.currentStatus.VFO = freq…

I do not recall if there is an InActiveRadioPtr. There is an ActiveRadioPtr.

Tom
Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

n4af, 2020-09-05:
well, my preference would be to test for band being same.
i think InactiveRadioPtr.BandMemory works in uBandMap.

73, Howie

n4af, 2020-09-06:
Added 4.92.1

In uBandMap procedure TuneRadioToSpot, I added the following after 
if (EntryBand = NoBand) then exit;
---------------------------------->
   if QSYInactiveRadio then
    if ((InActiveRadioPtr.BandMemory <> EntryBand) and (EntryBand = ActiveRadioPtr.BandMemory)) then
     begin
      QuickDisplay(TC_2radio_warn);
      exit;       // 4.92.1
     end;
  if not QSYInactiveRadio then
    if ((ActiveBand <> EntryBand) and (EntryBand = InActiveRadioPtr.BandMemory))  then        // 4.92.1
       begin
        QuickDisplay(TC_2radio_warn);
        exit;
       end;

TC_2radio_warn = 'ERROR 2 radios on same band'

n4af, 2020-09-06:
4.92.2 is out for testing. This adds an error msg if user has IN BAND LOCK set to TRUE (default) and attempts to select a band map entry that would result in both rigs being on the same band.

closed 2020-09-15 by n4af