TR4W/TR4W #452: With DIGITAL MODE ENABLE, CW no longer selectable

n4af, 2020-09-15:
Using 20M as an example and starting in CW mode on 14025, ALT-M switches to digital mode on frequency 14.225.
This should probably be 14075.
ALT-M then changes to SSB/14225.
Pressing ALT-M again should cycle back to CW 14.025
Frequency, indeed, does change to 14025 but Mode quickly goes to DIGITAL rather than CW.
So , there is no way to return to CW and frequency of 14.025 for digital is wrong.

ny4i, 2020-09-15:
There is a more general question if we should be changing the frequencies when a user switched the modes.

I ask as the modes are frequent wrong. Say I am on 7.060 LSB working split in a contest, there is no reason to change my frequency just because I change modes.

Should we have those be two separate things. Change mode or change frequency with spinning the dial or typing in the frequency?

Tom
Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

n4af, 2020-09-15:
Probably leave frequency for the end user.
The only drawback is cw to ssb where it a lot of knob turning. But, perhaps
modern rigs all have 'save' function for
freq and certainly clicking on a spot works.

ny4i, 2020-09-15:
Band stacking registers help and I always use the shortcut to type the frequency into the call window. So if I am on 14.025, I just enter 250 <enter> and the programs change the rig to 14.250. Very handy feature.

We might have a revolt on our hands though if we change the frequency mode memory feature without letting everyone know. :)

Tom
Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

n4af, 2020-09-15:
Yep, you could always put it up on the forum...

ny4i, 2020-09-15:
Am I right that the frequency that pops up is the last frequency you were on for that mode?

I seem to recall some code about a BANDMEMORY.

If I am right, that is a nice feature. The place we get in trouble is when there is nothing in the BANDMEMORY and the program sets it to a frequency in that mode’s neighborhood.

Tom
Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

n4af, 2020-09-15:
well, it appears it is hardwired to 14225 -or- 14.025

ok1rr, 2020-09-16:
4.92.3 - Direct clicking onto a spot in the Cluster window causes tuning 
of my K3 onto wild frequency out of any ham radio band. Clicking onto a 
bandmap entry works OK. In 4.92.0 can click either onto cluster entry or 
bandmap entry and no problem.

73,
Martin, OK1RR

n4af, 2020-09-16:
Hi Martin -

Could download again and try ?

Thanks, Howie

n4af, 2022-10-23:
no activity/feedback. closing.

closed 2022-10-23 by n4af