TR4W/TR4W #457: Cabrillo for RTTY shows mode incorrectly as DG

n4af, 2020-10-11:
I have worked TR4W+MMTTY in the Makrothen RTTY Contest.
In the Station Information window I chose RTTY,
but in cabrillo showed up as DG.

QSO: 14084 DG 2020-10-10 0543 UR7QM         KN76       R8XF       LP51
QSO: 14090 DG 2020-10-10 0546 UR7QM         KN76       IK7RVY     JN80
QSO: 14091 DG 2020-10-10 0547 UR7QM         KN76       UA9FLK     LO88

The log was not accepted by the sponsors:

ny4i, 2020-10-11:
When working the stations was the program in Digital or RTTY mode?

Principal Solutions Architect
Better Software Solutions, Inc. 
727-437-2771

n4af, 2020-10-11:
I do not know how one selects 'RTTY' mode ? I see only digital ??

n4af, 2020-10-11:
looking at the adif, most entries are flagged mode:7<DIGITAL with editable log showing '20NON' and producing NO Cabrillo.
The few entries showing MODE:4<RTTY format to correct 'RY' Cabrillo.
Almost like a hardware problem on what I think is a FT847.

ny4i, 2020-10-13:
My mistake. I got ahead of myself. That is in a branch.

Tom
Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

n4af, 2020-10-15:
This turned out to be inability of tr4w to address/reference the (valid) extmode coming back on polls in uRadioPolling.pas
Fixed 4.93.3

n4af, 2020-10-15:
i should have added that modifcation of extendedmode arrays was not necessary.
what was required was
RData.ExtMode := ActiveRadioptr^.CurrentStatus.ExtendedMode ;
in MainUnit.

ny4i, 2020-12-05:
I see when I reverted the ExtendedMode, that the mode is set as DIGITAL in the Mak contest. I am checking why that is since the extendedMode is RTTY. This is most likely an issue in PostUnit when we do the actual export as it happens on ADIF too.

ny4i, 2021-01-16:
Closed by FT4 and FT8 mode fixed and CAB in RTTY contest fix.

closed 2021-01-16 by ny4i