TR4W/TR4W #458: Bandmap blinking makes it hard to select a spot

n4af, 2020-11-09:
Under CW contest conditions, the spot flow is heavy enough that screen refresh can make it difficult to select a spot.

ny4i, 2020-11-09:
I researched all kinds of ways to get windows to stop blinking when it does updates. I could not figure out the Win32 dialog setting to do it.

I did notice that if I moved the Telnet window so I could not see it, it would settle down and stop flashing the screen as much but have not investigated that fully.

Tom
Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

n4af, 2020-11-09:
Hi Tom -

Try 4.94.2 <https://tr4w.net/4.94/tr4w_setup_4_94.2.exe> using DISPLAY
REFRESH from ctl-J

73, Howie

n4af, 2020-11-09:
A new parameter is added to CTL-J selections-
**DISPLAY REFRESH**. 
It can be set to refresh the bandmap after a range of 
0 = every spot from cluster window
9 - every 10 spots.

n4af, 2020-11-09:
Well, although this certainly helps with the blinking (refresh) situation it is not a solution.
 
The problem is that clicking on a spot can cause the rig to tune to a different spot- because the bandmap is not properly refreshed to sync with the spots buffer.

n4af, 2020-11-12:
I do not believe much can be done about this. If a BANDMAPPREVENTREFRESH is ordered the problem is that the spots still flowing in must be written to the spots buffer to prevent data (spots) loss.
BUT, this means the displayed band map is not synced to the spots buffer and clicking on a spot will often reference the wrong entry. This can be reproduced by single clicking on spot in the bandmap, waiting, and then clicking on the last entry in the bandmap. There is an excellent chance you will find you are displaying the spot in the buffer, not what is displayed.

n4af, 2020-11-13:
i have created 4.94.2 using parameter DISPLAY REFRESH to control how many spots are buffered before a write to band map occurs.
4.94.2 defaults to DISPLAY REFRESH =1, which operates without buffering.

closed 2020-11-13 by n4af