TR4W/TR4W #459: Extra characters show up after save backup log

ny4i, 2020-11-22:
After I use the ALT command to save the backup log, in the QuickDisplay, I see a few extra characters after the name of the file It looks like we are not delimiting the display properly with the length of the filename.

I attached a screenshot.
[image]

n4af, 2020-11-22:
hmm. not sure how to reproduce ? (encl)

73, Howie

ny4i, 2020-11-22:
I might have to try on my system. Could be length of name. That is reproducible if one sets up a similar structure. 

Tom

Principal Solutions Architect
Better Software Solutions, Inc. 
727-437-2771