TR4W/TR4W #463: In ARRl Sweepstakes, the ADI file export put the section name in the state.

ny4i, 2020-11-27:
In Sweepstakes, the export of an ADIF file has the received section as the state. This should either be left off or a simple conversion done to convert the sections to the state for proper exporting.

n4af, 2020-11-27:
That does not sound right.
When you start ARRL SS and are prompted for your Section, that should carry
into the Cbr.

So if I enter 'SFL' my cbr should show 'SFL'.
What happens if you edit your .cfg and change the My Section ?
Something like this ?
QSO: 14024 CW 2012-11-03 2100 N4AF          1    B 58 SFL K8MP       2    B
62 OH
QSO: 14024 CW 2012-11-03 2101 N4AF          2    B 58 SFL N4ZZ       2    U
57 TN
QSO: 14024 CW 2012-11-03 2102 N4AF          3    B 58 SFL N4RV       1    B
74 VA
QSO: 14024 CW 2012-11-03 2103 N4AF          4    B 58 SFL K9NA       1    Q
32 AL
QSO: 14024 CW 2012-11-03 2103 N4AF          5    B 58 SFL K4VV       3    U
51 VA
QSO: 21024 CW 2012-11-03 2104 N4AF          6    B 58 SFL VE5ZX      7    A
63 SK
QSO: 14024 CW 2012-11-03 2105 N4AF          7    B 58 SFL VA2SG      1    A
93 QC
QSO: 21027 CW 2012-11-03 2105 N4AF          8    B 58 SFL KE7X       7    U
59 MT
QSO: 21028 CW 2012-11-03 2106 N4AF          9    B 58 SFL WD0T       11   B
77 SD
QSO: 14024 CW 2012-11-03 2106 N4AF          10   B 58 SFL N3ZZ       7    B
53 SCV
QSO: 21029 CW 2012-11-03 2107 N4AF          11   B 58 SFL KB7Q       10   U
60 MT
QSO: 21030 CW 2012-11-03 2108 N4AF          12   B 58 SFL N3IQ       14   B
71 MDC
QSO: 21033 CW 2012-11-03 2108 N4AF          13   B 58 SFL K5GN       14   B
72 STX
QSO: 14024 CW 2012-11-03 2109 N4AF          14   B 58 SFL K0AD       9    A
61 MN
QSO: 14024 CW 2012-11-03 2110 N4AF          15   B 58 SFL K4FPF      6    U
59 VA
QSO: 14024 CW 2012-11-03 2111 N4AF          16   B 58 SFL N4WE       3    A
62 AL
QSO: 14024 CW 2012-11-03 2111 N4AF          17   B 58 SFL K0VBU      9    A
59 KS
QSO: 14024 CW 2012-11-03 2111 N4AF          18   B 58 SFL W9RE       22   B
62 IN
QSO: 14024 CW 2012-11-03 2112 N4AF          19   B 58 SFL VA3EC      6    A
88 GTA
QSO: 14024 CW 2012-11-03 2112 N4AF          20   B 58 SFL K9QC       2    A
75 WI
QSO: 14024 CW 2012-11-03 2114 N4AF          21   B 58 SFL WD1H       2    A
74 EM
QSO: 21034 CW 2012-11-03 2114 N4AF          22   B 58 SFL K0PY       13   A
92 KS

ny4i, 2020-11-27:
Right but I am talking about ADIF which is generate differently. I will check it out. NO doubt a regression I caused when I re-wrote the ADIF export.

ny4i, 2020-11-27:
Yes I found it. I will make a branch and fix it.

ny4i, 2020-11-27:
I fixed it. I will submit the branch as a pull request after I do a bit more testing tomorrow. I also handled VE providences I that map directly to a state. Note the STATE field in ADIF is really the primary administrative subdivision of a country. In the US, that means states, In Canada, that means Providences (again where they map naturally). MDC, VI and PR do not map to a state as in the MDC case, we cannot really know if it was MD or DC (not a state).

n4af, 2020-11-30:
Fixed in 4.94.4

closed 2020-12-09 by n4af