TR4W/TR4W #478: IC7600 fails to set frequency from bandmap

n4af, 2021-02-08:
Problem. Two owners of ICOM-7600 transceivers informed me that if you click
on the callsign on BANDMAP, it will be entered into the window,
but the transceiver will not switch to this frequency.
Although the CAT system is working fine.
Version 4.97.0 is installed by both owners.

ny4i, 2021-02-09:
Note the Icom 7600 had an upgrade option for the CI-V commands. We may need to know what firmware version the radio in question uses.

Steps to test this:

- [ ] Try on Icom IC-7300
- [ ] Verify the commands sent for the IC7600
- [ ] Check manual for commands to change frequency in IC7300 and IC7600 manuals

Icom 7600 uses the 5 byte frequency for the VFO for commands 00, 03 and 05.

**The question of the CI-V upgrade may be the issue. The code will use the Extended VFO commands for radios that support VFOB. The Icom 7600 is in that class if the firmware is upgraded. We need to know if the radio firmware is updated to at least version 2.00 (See https://www.icomjapan.com/support/firmware_driver/2387/) Anything short of version 2 (which came out in 2015) would be a problem.** 

If the user cannot upgrade for some reason, we would need to add an option to select non-CIV_Upgrade version but Iwould state that firmware from 2015 should be considered a minimum to use TR4W.

ny4i, 2021-02-13:
Email from UB3A

```
Hi -

Info UB3A:   I don’t use N1MM. Only TR4W.
My IC-7600 v.2.0.  See clip.

tr4w.log with TRACE
```

Part of the TRACE is below.

```
13 ôåâ 2021 19:39:12.341 56054 [10412] debug TR4WDebugLog  - [SetRadioFreq] set frequency to 3589500
13 ôåâ 2021 19:39:12.351 56064 [10412] debug TR4WDebugLog  - [PutCallToCallWindow] Putting US1IB into main call window
13 ôåâ 2021 19:39:12.625 56338 [9764] trace TR4WDebugLog  - >>>>Enter WriteToCATPort
13 ôåâ 2021 19:39:12.625 56338 [9764] trace TR4WDebugLog  - [Ðàäèî 1] Calling WriteFile to write 9 bytes <FEFE7AE026000002>
13 ôåâ 2021 19:39:12.767 56480 [9764] trace TR4WDebugLog  - >>>>Enter WriteToCATPort
13 ôåâ 2021 19:39:12.767 56480 [9764] trace TR4WDebugLog  - [Ðàäèî 1] Calling WriteFile to write 12 bytes <FEFE7AE025000095580300>
13 ôåâ 2021 19:39:13.781 57494 [9764] debug TR4WDebugLog  - [PutCallToCallWindow] Putting US1IB into main call window
13 ôåâ 2021 19:39:19.534 63247 [9764] debug TR4WDebugLog  - [PutCallToCallWindow] Putting OE2LCM into main call window
13 ôåâ 2021 19:39:20.688 64401 [9764] debug TR4WDebugLog  - [PutCallToCallWindow] Putting US1IB into main call window
13 ôåâ 2021 19:39:25.318 69031 [9764] debug TR4WDebugLog  - [PutCallToCallWindow] Putting US1IB into main call window
13 ôåâ 2021 19:39:26.497 70210 [9764] debug TR4WDebugLog  - Clearing main call window
13 ôåâ 2021 19:39:30.075 73788 [10412] debug TR4WDebugLog  - [SetRadioFreq] set frequency to 3518500
13 ôåâ 2021 19:39:30.083 73796 [10412] debug TR4WDebugLog  - [PutCallToCallWindow] Putting PA2W into main call window
13 ôåâ 2021 19:39:30.539 74252 [9764] trace TR4WDebugLog  - >>>>Enter WriteToCATPort
13 ôåâ 2021 19:39:30.539 74252 [9764] trace TR4WDebugLog  - [Ðàäèî 1] Calling WriteFile to write 9 bytes <FEFE7AE026000002>
13 ôåâ 2021 19:39:30.681 74394 [9764] trace TR4WDebugLog  - >>>>Enter WriteToCATPort
13 ôåâ 2021 19:39:30.681 74394 [9764] trace TR4WDebugLog  - [Ðàäèî 1] Calling WriteFile to write 12 bytes <FEFE7AE025000085510300>
```
[tr4w.log](https://github.com/n4af/TR4W/files/5976137/tr4w.log)

A couple of notable things I see...

- When trying to set the radio to frequency 3589500, we send a mode command of FEFE7AE026000002. The maps to LSB, Data mode off and Icom filter setting of 2 which is ok.

[image]

I then see where we are setting call (presumably from double clicking on the band map). But IO do not see a command got set the radio frequency in between the commands.

[PutCallToCallWindow] Putting US1IB into main call window
[PutCallToCallWindow] Putting OE2LCM into main call window
[PutCallToCallWindow] Putting US1IB into main call window
[PutCallToCallWindow] Putting US1IB into main call window 

The question therefore is why we did not set the frequency.

ny4i, 2021-02-23:
I added some more debug login in pull request #486. I cannot get it to fail on my IC7300.

ny4i, 2021-02-23:
Now in Pull request #487

n4af, 2021-02-23:
I am getting a rather nonsensical compile error in tr4w.dpr
[Fatal Error] tr4w.dpr(111): File not found: 'utils_text.dcu'

old and new tr4w.dpr have no change to them
utils_text.dcu last touched in 2018

argh..

ny4i, 2021-02-23:
Is your search path setup ok? Do as Build All maybe?

Tom
Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

n4af, 2021-02-23:
i tried commenting out most of the added code w/o success.
build all or build tr4w produces same result.
I will restore back to 4.97.7 and ensure compile is ok...

n4af, 2021-02-25:
lets keep this specific to IC7700/UY5ZZ and I will open new issue IC7600/UB3A.

n4af, 2021-02-25:
UY5ZZ has sent a trace under enhanced trace 4.97.8, but we are not clear whether the IC7700 was updated to firmware 2.2, which NY4I states is necessary.

n4af, 2021-02-25:
IC7700 moved to issue 488.

n4af, 2021-05-05:
inactive

closed 2021-05-05 by n4af