TR4W/TR4W #481: ADIF for PACC wrong

n4af, 2021-02-15:
4.90.10 introduced a problem with ADIF produced for ActiveExchange of RSTAndQSONumberorDomesticQTHExchange.
The first ADIF line should show STX_STRING of '1'. Instead the string is set to 'DX' and the next ADIF line for STX  shows qso number of '1'. All subsequent STX lines are off by one.

n4af, 2021-02-15:
I think this would happen for any AE of the above type where MyState is nil.

Module LogDupe.pas has been modified to add line
ExchangeInformation.QSONumber := TRUE;

4.97.5 is out for testing.

ny4i, 2021-02-16:
I can check this

Principal Solutions Architect
Better Software Solutions, Inc. 
727-437-2771

n4af, 2021-02-16:
Thanks Tom. Positive f/b on 4.97.5, so closing.

closed 2021-02-16 by n4af