TR4W/TR4W #482: Frequency polling with the Elecraft K2

SM6FPC, 2021-02-17:
Frequency polling the Elecraft K2 is eratic and is getting worse with versions above 4.96. With v4.86 Windows XP and the K2, the frequency is still logged correctly when the rig switches from RX to Tx, but not displayed continuosly. I also see a warning "Operating Split" now and then. With the 4.97.1 version I have to force a band switch by pushing the K2 bandswitch buttons to update the frequency. The Auto CQ may restart in the middle of the previuos message and now and then skips a CQ message (long pauses). I have found a workaround by using the LP-bridge software by N8LP. From the LPB2 manual: "LPB2 actually intercepts and buffers the additional polling traffic and responds immediately without sending the additional polling commands to the rig, which reduces serial traffic and eliminates packet collisions."  The LPB2 is transparent to other commands so "CW by CAT" is OK.
Letting LPB2 do the polling solves all the problems described above with any version of TR4W. Even the automatic swith from CQ to S&P mode when the frequency changes now works perfectly. There is one drawback. The LPB2 will not run under Windows 10. (for LPB2 see www.telepostinc.com) With Windows 10 and TR4W 4.96 without LPB2 I see the same problems but switching the rig from Rx to Tx still updates the frequency, so the log is acceptable. 
With my old IC725 I see no polling problems. Shortening the poll time to 10ms gives a few more correct readings with the K2. I can see the polling request in the LPB2 software window: 10 times "IF;" then one "FB;" command, looks OK to me. I noted a polling issue with the K3 from 11 Dec 2020. (Peter SM6FPC)

n4af, 2021-02-17:
Hello Peter -

I am not able to reproduce that.
Hundreds of qsos a week on two k3's in so2r on CW using Windows 10 and 4.97

73, Howie N4AF

SM6FPC, 2021-02-17:
I am talking about the K2! Maybe the K3 is different.I have used TR4W for many years and with 2 different laptops, one old with XP and a new with Windows 10. The problem is the same with both laptops and is solved by running the LPB2 in the XP laptop.73, Peter SM6FPC Skickades från E-post för Windows 10 av Peter van Haeringen Från: Howie HoytSkickat: den 17 februari 2021 16:49Till: n4af/TR4WKopia: SM6FPC; AuthorÄmne: Re: [n4af/TR4W] Frequency polling with the Elecraft K2 (#482) Hello Peter -I am not able to reproduce that.Hundreds of qsos a week on two k3's in so2r on CW using Windows 10 and 4.9773, Howie N4AF

SM6FPC, 2021-02-17:
Just Another thought. Could it be the USB to serial interface is too slow? I use an old PL2303HXA chip. Skickades från E-post för Windows 10 av Peter van Haeringen Från: Howie HoytSkickat: den 17 februari 2021 16:49Till: n4af/TR4WKopia: SM6FPC; AuthorÄmne: Re: [n4af/TR4W] Frequency polling with the Elecraft K2 (#482) Hello Peter -I am not able to reproduce that.Hundreds of qsos a week on two k3's in so2r on CW using Windows 10 and 4.9773, Howie N4AF

n4af, 2021-02-17:
This is very difficult to read.
Yes, certainly any problem related to CAT speed would be problematical.

73, Howie

SM6FPC, 2021-02-17:
I just checked the Elecraft forum for ”K2 and Prolific”. There are some warnings for old Prolific chipsets (like mine) in these threads. I Think I will buy a new non Prolific USB to serial. I will comment it on the GitHub issue when I see some results useful to other K2 owners.Thank’s for putting me on this trail. Always helpful to discus with others. Skickades från E-post för Windows 10 av Peter van Haeringen Från: Howie HoytSkickat: den 17 februari 2021 17:52Till: n4af/TR4WKopia: SM6FPC; AuthorÄmne: Re: [n4af/TR4W] Frequency polling with the Elecraft K2 (#482) This is very difficult to read.Yes, certainly any problem related to CAT speed would be problematical.73, Howie

ny4i, 2021-02-17:
Prolific usb interfaces should be thrown in the trash. They are not reliable. 

Principal Solutions Architect
Better Software Solutions, Inc. 
727-437-2771

SM6FPC, 2021-02-19:
Yes, the above got I little bit hard to read...
Howie wrote more diplomatically: "Yes, certainly any problem related to CAT speed would be problematical."
I just bought and installed a new USB to Serial. Now TR4W runs on both laptops in all versions without the LP-bridge software. Problem solved!! TU Howie and Tom for putting me on the right track!

closed 2021-02-19 by SM6FPC