TR4W/TR4W #488: IC7700 fails to set frequency from bandmap

n4af, 2021-02-25:
This is connected to issue 478 on ic7600 but is IC7700 from UY5ZZ.
Present status is we have a 4.97.8 trace from UY5ZZ but it is NOT clear that the IC7700 has been updated to current firmware 2.2
NY4I insists this needs to be done for set frequency to work correctly.

n4af, 2021-02-25:
NY4I has been unable to recreate #478 or #488 via his IC7300.
He updated/enhanced trace to 4.97.8 which we need -if- firmware at 2.2 ???

n4af, 2021-02-25:
`25 ôåâ 2021 12:00:33.687 117156 [2624] trace TR4WDebugLog  - [BandMap::BandmapDlgProc] Calling TuneRadioToSpot for active Radio
25 ôåâ 2021 12:00:33.687 117156 [2624] trace TR4WDebugLog  - Entering TuneRadioToSpot IK4ALM - 14005000
25 ôåâ 2021 12:00:33.687 117156 [2624] trace TR4WDebugLog  - [TuneRadioToSpot] Calling SetRadioFreq 14005000
25 ôåâ 2021 12:00:33.687 117156 [2624] debug TR4WDebugLog  - [SetRadioFreq] set frequency to 14005000
25 ôåâ 2021 12:00:33.687 117156 [2624] debug TR4WDebugLog  - [PutCallToCallWindow] Putting IK4ALM into main call window
25 ôåâ 2021 12:00:33.796 117265 [1408] trace TR4WDebugLog  - [ReadFromSerialPort] Read þþtà ýþþàt  ý                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                  from serial port
25 ôåâ 2021 12:00:33.796 117265 [1408] trace TR4WDebugLog  - >>>>Enter WriteToCATPort
25 ôåâ 2021 12:00:33.796 117265 [1408] trace TR4WDebugLog  - [Ðàäèî 1] Calling WriteFile to write 9 bytes <FEFE74E026000002>
25 ôåâ 2021 12:00:33.921 117390 [1408] trace TR4WDebugLog  - [ReadFromSerialPort] Read þþtà&  ýþþàtúý                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                  from serial port
25 ôåâ 2021 12:00:33.921 117390 [1408] trace TR4WDebugLog  - >>>>Enter WriteToCATPort
25 ôåâ 2021 12:00:33.921 117390 [1408] trace TR4WDebugLog  - [Ðàäèî 1] Calling WriteFile to write 12 bytes <FEFE74E025000050001400>
25 ôåâ 2021 12:00:34.046 117515`

ny4i, 2021-02-25:
Thanks. Embedded inside that data (I looked at it in Hex with Notepad++) is an NG response code from Icom 7700. This means it does not understand the 25 and 26 Extended Band and VFO commands. 

Please update the 7700 to firmware version 2.1 or higher and test again. I cannot do anything else until the radio is updated. That firmware came out in the year 2015 so 6 years is a bit long to wait and in my view, perfectly reasonable to ask program users to upgrade.

https://www.icomjapan.com/support/firmware_driver/1939/

n4af, 2021-05-05:
inactive

closed 2021-05-05 by n4af