TR4W/TR4W #489: TR4W fails to switch rig MODE`

n4af, 2021-02-28:
When using Alt-M to change modes from CW to SSB on my K3, TR4W indicates SSB and moves the frequency to the phone band but does not change the mode on the radio.

TR4W logging indicates the mode change but the rig does not actually change mode.

ny4i, 2021-03-01:
Howie - Are you looking at this or do you want me to fix it? I did not want to step on any work.

n4af, 2021-03-01:
Hi Tom -

You are probably better at this:
between
LOGWIND proc DisplayBandMode
produces the EXISTING freq to qsy to.
Eg, if I R on 3525khz and press altm,
LogRadio rt.Kenwood gets order to qsy to 3525 (instead of 3850).
I would think that we have a bad FreqMemory array in VC.pas ----> but do
not see it ???

Tks, Howie
 i

n4af, 2021-03-01:
oh, one other thing.
This problem starts on 4.91.10
https://github.com/n4af/TR4W/tree/4.91.10/tr4w/src/trdos

n4af, 2021-03-01:
well, per the url for 4.91.10,
https://github.com/n4af/TR4W/commit/b469ba866700c5bcbf1aecbc29411dfb4704250a#diff-47072cd2f3fcd8d900ab0195533a4b9329b5b6321111703648cf2637bdac89e0L2741-R2759
looking at the change if I revert, the mode changes correctly.
The freq still stays tho...

adding this removed code makes mode change correctly:
tKenwood14Bytes[0] := 'M';
tKenwood14Bytes[1] := 'D';
tKenwood14Bytes[2] := CHR(TempByte);
tKenwood14Bytes[3] := ';';
AddToOutputBuffer {WriteToCATPort}(tKenwood14Bytes, 4);

begin
if ActiveRadioPtr.CurrentStatus.Mode <> Digital then
begin
tKenwood14Bytes[0] := 'M';
tKenwood14Bytes[1] := 'D';
tKenwood14Bytes[2] := CHR(TempByte);
tKenwood14Bytes[3] := ';';
AddToOutputBuffer {WriteToCATPort}(tKenwood14Bytes, 4);
end
else
begin
logger.debug('Skipping Kenwod mode change because MODE was already DIGITAL'
);
end;
end;

ny4i, 2021-03-01:
OK. I will look at it tonight.

Tom
Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

closed 2021-03-06 by n4af