TR4W/TR4W #495: Filter bandmap by entries in Initial exchange file

n4af, 2021-03-19:
Please add the ability to only write out bandmap entries that match content of the Initial Exchange file.

n4af, 2021-03-19:
A new configuration command will be added-
IE SWITCH
When set to 'TRUE', only spots matching entries in the active initial exchange file will be written out to the bandmap.

n4af, 2021-03-19:
4.98.8

ny4i, 2021-03-19:
What is an example of something in the Initial Exchange file that I could limit? By call, location, mode, etc?

Tom
Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

n4af, 2021-03-19:
No, only by call.
So if the call is in the IE, a spot for that call should make it into the bm.

n4af, 2021-05-05:
i have used this many times in CWT.

closed 2021-05-05 by n4af

n4af, 2021-08-12:
note: ie_switch searches BOTH the initial.ex AND TRmaster.dta files