TR4W/TR4W #498: Option to make FM a separate mode from SSB

n4af, 2021-04-08:
In some local VHF contests, CW, SSB and FM are considered different modes.
But TR4W also defines FM = SSB and accordingly defines it as Dupe.

Is it possible to make FM and SSB as different modes, suppose,
if CONTEST TITLE = VHF is written in the config?
Or enter a new command in CTRL + J:
FM EQUALS SSB = TRUE (FALSE)?

ny4i, 2021-04-08:
Unless you are doing this, I can take a shot. 

Tom

Principal Solutions Architect
Better Software Solutions, Inc. 
727-437-2771

n4af, 2021-04-08:
THANKS Tom !

73, Howie

ny4i, 2021-04-17:
I poked around the code a bit to get a feel for the effort. FM is one of the standard modes, but in many places, if the mode = FM, it is changed to Phone. The extendedMode parameter I added for WSJT-X helps us here as I have in extendedMode of eFM. So even if the mode is FM, the actual mode can be checked in extendedMode. 

The first place I can think this needs to be be changed is in how we determine Multipliers. I know Valery suggested FM EQUALS SSB = TRUE (FALSE) but the question is if that is clear enough that we should consider FM and SSB separate mults. Once can presume that if a contest allows points for both FM and SSB, then they would consider that as two different MULTS.

This is why I really need to see the rules for the contest in question. We would need a FM Equals SSB for points and FM Equals SSB for points. bDistinctFMSSB_Mults and bDistinctFMSSB_points.

Then all the code that checks for mults has to be updated to look for that and treat prospective contact accordingly.

Same goes for the scoring calculations to look at them separately.

closed 2021-05-05 by n4af