TR4W/TR4W #502: TR4WSERVER not able to synchronize logs between server and clients

OH6QU, 2021-04-16:
TR4WSERVER is not able to syncronize logs between server and clients in TR4W v4.99.0. It is still working in v4.84.0. I assume it has something to do with a new format in the TR4W binary formatted log files.

Jari OH6QU

ny4i, 2021-04-17:
You mean when you are running in network mode using the TR4WSERVER as a central server?

All clients are using a recent version of TR4W, correct?

Tom
Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

OH6QU, 2021-04-17:
Tom, yes. I installed TR4W v4.99.0 into two separate PCs. One of them was running TR4WSERVER + TR4W client. The other PC was running TR4W client only. I disconnected an ethernet cable to the latter PC and then logged some QSOs. Then I plugged the ethernet cable back. The TR4W client did recognize that TR4WSERVER has more QSOs and it tried to synchonize but that failed. I repeated same procedure with TR4W v4.84.0 clients and everything was working OK.

Jari OH6QU

ny4i, 2021-04-17:
After looking at the project, I believe the issue is the current TR4WSERVER is not being included in the release.

Howie - Can you compile the TR4WSERVER project before updating the distribution?

That should take care of the issue.

n4af, 2021-04-17:
i cannot get tr4wserver to compile. suspecting directories setup wrong...

ny4i, 2021-04-17:
I noticed that. It looked like ..\tr4w\src was causing an issue for TL. That is the right directory, isn’t it?

Tom
Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

n4af, 2021-04-17:
right now i fail in tree.pas.
[Error] Tree.pas(22): Invalid compiler directive: 'MODE'
on line 22
followed by
[Fatal Error] Tree.pas(39): File not found: 'FileUtil.dcu'
[Error] Tree.pas(840): Undeclared identifier: 'SYSTEMTIME'
[Error] Tree.pas(852): Undeclared identifier: 'DWORD'
[Error] Tree.pas(853): Undeclared identifier: 'HWND'
[Fatal Error] uNet.pas(32): Could not compile used unit 'utils_net.pas'

n4af, 2021-04-17:
OK, I  have a good compile of both tr4w and server.
Please test [4.99.4](https://tr4w.net/4.99/tr4w_setup_4_99.4.exe) and see if server being back-level was the problem.

Thanks, Howie

ny4i, 2021-04-17:
Howie,

I did a fresh install (even deleted tr4wserver.exe) but the file is still dated 2014. It is 18944 bytes. Can you check what went into the build? Note I did confirm in the details of the install that it did copy tr4wserver.exe

When I run it, it is version 1.41.

[image]

Thanks,

Tom

n4af, 2021-04-18:
Yes, another problem.
Pls try fresh install 4.99.4 <https://tr4w.net/4.99/tr4w_setup_4_99.4.exe>
again.

tks, h

ny4i, 2021-04-18:
The QSOs created are going across but when I do a synchronize, I get all error about Failed to receive server log.

I will take a look.

ny4i, 2021-04-18:
Howie - Please check in what you changed to MASTER or make a branch so I can grab that to test this error. Thanks.

n4af, 2021-04-18:
Hi Tom -

I have pushed 4.99.4 onto the repo.

Tks, Howie

n4af, 2021-04-18:
Jari -
Curious how you are using tr4wserver ? 
Is it 2BSIQ ?

73, Howie

OH6QU, 2021-04-18:
Howie, my station isn't ready for SO2R / 2BSIQ yet but that is the ultimate goal I am pushing towards. I have all necessary components to build it but the work is still in progress.

The reason I was testing TR4W with TR4WSERVER is the IARU HF Contest in July. There is a class for HQ-stations in this contest. We are going to setup 15-20 stations around the Finland which are representing Finnish Radio Amateur League in this contest using the same callsign in different bands and modes. For that purpose we need a robust software that has networking capabilities and can handle that many clients. 

Last year we were using other logging software and it ended up in a complete disaster. The software started to crash after 30 mins and time-to-time we were forced to do logging by pen & paper. As you may guess this hassle with the logging software was reducing our score in the contest a lot as we didn't know the frequencies of other bands & modes. Also during the contest the band map didn't work as DXCluster was causing too much load to that logging software.

Jari OH6QU

n4af, 2021-04-19:
testing seems to indicate this started at 4.90.10

ny4i, 2021-04-21:
Howie - Sorry as I have not returned to this. Work has been busy (busy week). I was thinking the best way to track this down is to check int he server what length message it is sending and see compare it to that line of code that was calculating the length and reporting the error. In the main log program logger.debug('Log record length = %d',[nLen]); % Where you put the result of the addition in nLen. That would be the first place to start.

What makes no sense is the addition should be derived from the size of the records as nothing appeared to be hardcoded. Unless the Write in the Server is hardcoded.

n4af, 2021-04-22:
Hi Tom -
I played with it earlier today.
90.10+ deviation is on size of header and exchange (=300)  ---> 1024 - 300
mod 300  (90.10+ failure)
on 90.9    (1024 - 256) mod 256

Previously < 90.9, the above were both 256.

when it works you have (1024 - 300)  mod 300 ---> A-OK
ie, totalbytes minus the header must be divideable by the sizeof the
contestexchange
or, put another way, the TotalBytes (- header)  MUST be an exact multiple
of the size of the exch.
when it fails, you have 724 mod 300 in uGetLogServer
 if (LogSize <> TotalBytes) or ((TotalBytes - *SizeOfTLogHeader*) mod
*SizeOf(ContestExchange)* <> 0) then

If that does not help I can send you the source for 4.90.8.
but i suspect your comments below cover the problem ?
(* The TLogHeader type had a dependency on the size of the ContestExchange
   to build the dummy area of its type definition. This meant that to add
something
   to ContestExchange, you had to go to find TLogRecord to change the
number of
   bytes in ther dummy array. I changed this (at bottom of VC.pas) so the
size
   is derived on the sizes so no change elsewhere required.
   I found this because I added a field to CE and the program complained as
the
   TRN file was not the right size. This should eliminate that maintenance
   step. de NY4I
*)

Does that make sense or would you like the 90.8 source-
So you can step thru it as I did above ?

Tks, Howie

n4af, 2021-04-22:
sorry, correction follows:

works:
1024-256/256
fails:
1024-300/300
so *SizeOfTLogHeader*) *SizeOf(ContestExchange) jumped by 44 bytes*

ny4i, 2021-04-22:
That’s quite complex. I wonder why the multiple of 256 is even a factor? Thanks for pointing the way. I’ll look at it further. 

It might be using it as a basic validation. 

Tom

Principal Solutions Architect
Better Software Solutions, Inc. 
727-437-2771

n4af, 2021-04-22:
i think he is just validating that the total log size (minus header) is an
exact multiple of the exch records.
might be interesting to comment that out and see works ok - or corrupts
data ;-)

73, Howie

ny4i, 2021-04-22:
Right but it only applies if the Contest Exchange is a multiple of 256 bytes if I understood what you wrote correctly.

Tom
Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

n4af, 2021-04-22:
Yes, in this case size of exch and header should be 256.
It would help if I copy in the 'difference in logs' screen and the
(clobbered) result using 300.

update: perhaps best way to describe-->
the local log has been bumped to records of 300 bytes (after 4.90.9)
WHILST
*THE SERVER LOG IS STILL USING RECORD LENGTH OF 256 !!!*
ref attachments...
#1 trying to copy 300b local log into 256b server log
#2 actual log sizes. i confirm this is what displays on respective file
properties

ny4i, 2021-04-26:
Tough problem. I spent a couple of hours on this tonight. Still looking to understand it. I was looking at the logger but let me check for a size of 256 in the server. I thought that was derived from the sizeof(ContestExchange) so no sure why server log would not be 300 too.

n4af, 2021-04-26:
I have also  been looking at this w/o much success:
 if OpenServerLog(OPEN_ALWAYS) then
          if ServerLogHandle <> INVALID_HANDLE_VALUE then
          begin
            if (Windows.GetFileSize(ServerLogHandle, nil) mod
SizeOf(ContestExchange)) <> 0 then
This test passes for all the wrong reasons--> comparing 512 to mod 256...

n4af, 2021-04-26:
Where is ContestExchangev1_5 declared to tr4wserver project ?

ny4i, 2021-04-26:
It’s not unless it picks up VC.pas. 

It should not need that however. 

Principal Solutions Architect
Better Software Solutions, Inc. 
727-437-2771

ny4i, 2021-04-27:
I think I figured it out.

It has to do with the serverlog.trw file in the tr4wserver folder. **First delete that.**

Then go remove all your DCUs. Or do a batch compile where you are sure the DCUs will be overwritten. **Do a BUILD of TR4W.**

Now open the TR4wSERVER. You may have some missing files. Go into Tools..Environment and add the proper path on your system to point to these directories:

c:\tr4w\tr4w\src
c:\TR4W\tr4w\src\trdos
c:\TR4W\tr4w\src\utils

I use c:\tr4w but you can change it to d:\newsrc.

Now rebuild and run tr4wserver (make sure you removed the serverlog.trw file.

After you run TR4wSERVER, the TR4WServer.trw file should have v1.6 in the first line. If not (and it shows v1.5), you still have the old file or an old DCU.

See if that works.

ny4i, 2021-04-27:
I should have said it is working here now.

n4af, 2021-04-27:
Hi Tom

> I think I figured it out.
>
> It has to do with the serverlog.trw file in the tr4wserver folder. *First
> delete that.*
done

> Then go remove all your DCUs. Or do a batch compile where you are sure the
> DCUs will be overwritten. *Do a BUILD of TR4W.*
done

> Now open the TR4wSERVER. You may have some missing files. Go into
> Tools..Environment and add the proper path on your system to point to these
> directories:
>
> c:\tr4w\tr4w\src
> c:\TR4W\tr4w\src\trdos
> c:\TR4W\tr4w\src\utils
>
> I use c:\tr4w but you can change it to d:\newsrc.
Looks like i need to have the include libs also

> Now rebuild and run tr4wserver (make sure you removed the serverlog.trw
> file.
I (think) serverlog.trw and dcu's are dynamically overlaid but I followed
your recommendation.
After I (FINALLY) got a good compile, I still had the conflict of running
comipiiled pgm- whilst IDE debug was working. very  encouraged to see
compile wrote out a good 300byte serverlog.trw (1.6) !

I think I have a working/compiled system, but want to experiment with it
some more in the morning.
SO NICE TO SEE SERVERLOG.TRW AT 300 BYTES !

BRILLIANT work Tom- once I read your update and realized there had to be a
problem with lib declarations or residual dcu the rest was easier ;-)

Thanks, Howie

n4af, 2021-04-27:
4.99.5 out for testing

ny4i, 2021-05-02:
Jari - Can you verify the latest version fixed networking?

I am also curious about the number of different stations. Do you plan to have 15-20 stations connected to one server via VPN or similar? 26 would be the limit as the stations are identified by a single alphabet letter (English). I am certain no one has ever tried to link 20 stations together to one server.

n4af, 2021-05-05:
closing.
There was a problem with port 1061/1062 being closed.
Otherwise all seems ok

closed 2022-02-08 by ny4i