TR4W/TR4W #503: In Florida QSO Party, Monroe county shows as MT

ny4i, 2021-04-25:
When entering MON for the county in the FQP, it shows up a MT (Montana the state). I suspect this is because the florida.dom file has INCLUDE FILE S50.DOM as the first DOM referenced. That brings in the s48.dom  (through s50.com) DOM which has an abbreviation for MON to be Montana (Mt = MT, MON,^Montana). If I move the INCLUDE FILE S50.DOM to the end of the florida.dom file, it accepts MON properly as Monroe county. 

So in QSO parties, should the INCLUDE FILE S50.DOM line always be at the end of the file?

I checked and with the INCLUDE FILE S50.DOM last if I work a station in MO, (Missouri), it shows properly. 

So it does seems like the INCLUDE FILE S50.DOM is the way to go in QSO parties (at least Florida but for any cases where we have a conflict between an equivalent state abbreviation that matches a county.

n4af, 2021-04-26:
The way FContest functions is that the CTY dom  array is concatenated off
of the contest array (in this case FLORIDA.DOM).

The simplest solution would be to remove the three letter designators (only
a few) off of P48.
So, 'MON' is removed and S48:
Mt = MT, Montana

This only occurs using IN-STATE.

n4af, 2021-04-27:
4.99.6 is available for testing.

ny4i, 2021-05-02:
This can be closed

closed 2021-05-02 by ny4i