TR4W/TR4W #505: In S&P mode, after sending call by hitting ENTER, F1 does not send call again

ny4i, 2021-04-25:
The scenario is as follows:

Program in S&P mode (indicated by green background in exchange window)

Enter a callsign of a station we want to call and hit enter.

Our call is sent.

If station goes back to someone else, we wait and when our turn, hit F1 again to send out call.

Nothing happens.

This is in CWBYCAT mode. The second line on K3 flashes as if it is doing something though. 

I have not tried it with a WinKeyer.

n4af, 2021-04-26:
This may be specific to CWBYCAT as WK responds to F1 in S&P continuously.

ny4i, 2026-11-04:
Tracked in the cross-platform TR4W (TR4W/TR4W-D12) under #66 (modify the CALL sent in F1 S&P) and #61 (S&P entered after F1 on Icom in split). Closing here to keep one place.

closed 2026-11-04 by ny4i