TR4W/TR4W #507: State QP

n4af, 2021-05-02:
Allow two entries for two counties. 
Portables park on county line and give out two or more.

n4af, 2021-05-02:
For Active Exchange = RSTDOMESTIC,
parse for two possible alpha county entries, negating CALLSIGN UPDATE ENABLE on the basis of all alpha.

n4af, 2021-05-13:
4.100.7 out for testing

ny4i, 2021-05-13:
Does anyone know for certain that the contest log checking software understands this is not a question of dupes or busted exchanges? I assume by now that all the contest log checking tools understand they can have the same station with different mults.

And speaking of which, if TR sees a new multi, should it count the points? I know that is minor since the log checking will calculate it but guys do love their 3830Scores posts.

I am talking more about since sate QSO parties which this feature can also be used.

Tom
Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

n4af, 2021-05-13:
Hi Tom -

> Does anyone know for certain that the contest log checking software
> understands this is not a question of dupes or busted exchanges? I assume
> by now that all the contest log checking tools understand they can have the
> same station with different mults.
I can confirm that INQP/N9LF states:

"When an Indiana Station sends two counties for INQP, this is counted as
two QSOs for scoring purposes. There are two ways to handle this situation.
You may log this as a single QSO by noting the exchange for both counties.
Alternatively, two QSOs can be logged, one for each county.

The Cabrillo file should look like this:
QSO: 7000 CW 02-MAY-2021 1600 N4AF 599 TN N9LF 599 INTPP INCAR

or

QSO: 7000 CW 02-MAY-2021 1600 N4AF 599 TN N9LF 599 INTPP
QSO: 7000 CW 02-MAY-2021 1600 N4AF 599 TN N9LF 599 INCAR

Either way, the log checking software counts two QSOs.
As for the points, the log checking will calculate the proper score for
each QSO and count the multipliers appropriately.

"
TR4W chose the latter example...

>
> And speaking of which, if TR sees a new multi, should it count the points?
> I know that is minor since the log checking will calculate it but guys do
> love their 3830Scores posts.
Preferably 'yes'. On the multi county version 4.100, all counties entered
are checked for new multiplier but, alas, only the first line of the
generated multi-lines is given qso points.
We could of generated lines for 'ny4i', 'ny4i/p' and (potentially) 'ny4i/r'
so that dupe checking would not flag the extra lines as 'dupe'  bit it as
decided to keep it KISS-
which also matches N9LF comments.
So presently, in p.m. to n4tz :
Known limitations :
1) *Can only be used for rst/domestic qth contests*
2)* Can only be used if S&P mode*
3) *Can only be used with NO RST entered (default will generate)*
4) *Rather than error on invalid county, it will be tossed*.
    example: INQP, I enter exchanges of 'inada inbar inbin'
the first two should generate lines,, 'inbin' will not generate an entry.
---
Howie

ny4i, 2022-02-08:
Is this working? Checking if this can be closed.

n4af, 2022-02-08:
Yes. I will close this.

closed 2022-02-08 by n4af