TR4W/TR4W #510: Update UDP records to match latest N1MM UDP specification

ny4i, 2021-05-02:
N1MM has made some changes to the UDP records. Check that TR4W is sending the same fields on all record types. Add any new records too.

Specifications are available here: https://n1mmwp.hamdocs.com/appendices/external-udp-broadcasts/

n4af, 2021-05-24:
uploaded as 4.100.10

ny4i, 2021-05-24:
I may have broke something here in the program changing bands. It sees the frequency ok but is not changing th ebands. This release should not go anywhere yet.

ok1rr, 2021-05-24:
Today tried MWC contest again. A "7" multiplier appeared. There are NO 
numerals as mults, a mult must be A to Z (max. 26, per band).

73,
Martin, OK1RR

n4af, 2021-05-25:
I am confused are you commenting on the UDP update of 4.100.10 or an
entirely new problem re: release 4.100.??

73, Howie

n4af, 2021-05-25:
rr.

ny4i, 2021-05-25:
Yep I broke something. Can you blackout that branch?

That is safest thing.

Tom
Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

n4af, 2021-05-25:
ok. i have reset the head back two commits and then
re-commited subsequent maintenance, so (hopefully) 4.100.10 code is gone.

ny4i, 2021-07-03:
This was fixed along the way

closed 2021-07-03 by ny4i